YosysHQ/nextpnr #467: Build error on opensuse leap

marph91, 2020-07-05:
I'm using opensuse leap and tried to build nextpnr from sources (commit 137241cfef385a68d2ce732a5b3283458790543d). Eigen 3.3.4 is installed. The build fails with the following output:
```bash
[ 68%] Building CXX object CMakeFiles/nextpnr-ice40.dir/common/placer_heap.cc.o
/home/martin/dev/nextpnr/common/placer_heap.cc:37:10: fatal error: Eigen/Core: Datei oder Verzeichnis nicht gefunden
 #include <Eigen/Core>
          ^~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/nextpnr-ice40.dir/build.make:197: CMakeFiles/nextpnr-ice40.dir/common/placer_heap.cc.o] Fehler 1
make[1]: *** [CMakeFiles/Makefile2:226: CMakeFiles/nextpnr-ice40.dir/all] Fehler 2
make: *** [Makefile:147: all] Fehler 2
```

My workaround was to create a symbolic link `sudo ln -s /usr/include/eigen3/Eigen /usr/include/Eigen` as described [here](https://github.com/opencv/opencv/issues/14868). Another workaround would be to use `# include <eigen3/Eigen/Core>` instead of `# include <Eigen/Core>`.

I guess it works on other linux distributions, since I didn't find any related bug report. Is there a way to fix this for all distributions?

daveshah1, 2020-07-05:
Seems like the CMake find_package Eigen3 isn't working properly for some reason. Out of curiosity, what does `pkg-config --cflags eigen3` show (I don't think cmake actually uses this but I would like to see what path it thinks it is installed to.)

marph91, 2020-07-06:
```bash
> pkg-config --cflags eigen3
-I/usr/include/eigen3
```

daveshah1, 2020-07-06:
Does `/usr/share/cmake/Modules/FindEigen3.cmake` exist? Are you sure that you are using system cmake and not, for example, Conda-provided cmake? This is certainly a mysterious failure.

marph91, 2020-07-06:
The file `/usr/share/cmake/Modules/FindEigen3.cmake` exists and looks correct.

I only have the system python installed. Anaconda or similar environments aren't installed.

I will try to search a bit more. Maybe this is a known problem with eigen3 + cmake + opensuse.

marph91, 2020-12-11:
Small update: There is a similar issue at openscad/openscad#2082. This points to the [eigen source code](https://gitlab.com/libeigen/eigen/-/blob/master/cmake/Eigen3Config.cmake.in). It seems like the `EIGEN_*` variables are legacy and shouldn't be used anymore.

Lunaphied, 2026-09-16:
This has been addressed in one of the later CMake revisions

closed 2026-09-16 by Lunaphied