YosysHQ/nextpnr #476: Cannot automatically find pytrellis in /usr/local/lib64

tdaede, 2020-07-22:
Configure correctly identifies /usr/local as the trellis prefix but then can't find trellis in /usr/local/lib64 without manually TRELLIS_LIBDIR.

whitequark, 2020-07-22:
Can you post the CMake output?

mossmann, 2021-02-23:
I encountered the same issue. Here's my cmake output:
```
mossmann@antigonus /tmp/nextpnr $ cmake . -DARCH=ecp5 -DTRELLIS_INSTALL_PREFIX=/usr/local
-- The CXX compiler identification is GNU 9.3.0
-- The C compiler identification is GNU 9.3.0
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.7.8", minimum required is "3.5") 
-- Found PythonLibs: /usr/lib64/libpython3.7m.so (found suitable version "3.7.8", minimum required is "3.5") 
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found Boost: /usr/include (found version "1.72.0") found components: filesystem program_options iostreams system thread regex chrono date_time atomic 
-- Found Boost: /usr/include (found version "1.72.0") found components: program_options filesystem system 
-- Check if the system is big endian
-- Searching 16 bit integer
-- Looking for sys/types.h
-- Looking for sys/types.h - found
-- Looking for stdint.h
-- Looking for stdint.h - found
-- Looking for stddef.h
-- Looking for stddef.h - found
-- Check size of unsigned short
-- Check size of unsigned short - done
-- Using unsigned short
-- Check if the system is big endian - little endian
-- Configuring architecture: ecp5
-- Enabled ECP5 devices: 25k;45k;85k
-- Trellis install prefix: /usr/local
-- Searching for pytrellis in: /usr/local/lib;/usr/lib;//lib;/usr/lib;/usr/local/lib;/usr/X11R6/lib;/usr/pkg/lib;/opt/lib;/usr/x86_64-pc-linux-gnu/lib//gcc;/usr/x86_64-pc-linux-gnu/lib/;/usr/lib64;/usr/libx32;/usr/lib32;/usr/lib;/lib
CMake Error at ecp5/CMakeLists.txt:64 (message):
  Failed to locate the pytrellis library

-- Configuring incomplete, errors occurred!
See also "/tmp/nextpnr/CMakeFiles/CMakeOutput.log".
See also "/tmp/nextpnr/CMakeFiles/CMakeError.log".
mossmann@antigonus /tmp/nextpnr $ cmake . -DARCH=ecp5 -DTRELLIS_INSTALL_PREFIX=/usr/local -DTRELLIS_LIBDIR=/usr/local/lib64
-- Found Boost: /usr/include (found version "1.72.0") found components: filesystem program_options iostreams system thread regex chrono date_time atomic 
-- Found Boost: /usr/include (found version "1.72.0") found components: program_options filesystem system 
-- Configuring architecture: ecp5
-- Enabled ECP5 devices: 25k;45k;85k
-- Trellis install prefix: /usr/local
-- Trellis library directory: /usr/local/lib64
-- Trellis data directory: /usr/local/share/trellis
-- Using ECP5 chipdb: /tmp/nextpnr/ecp5/chipdb
-- Configuring done
-- Generating done
-- Build files have been written to: /tmp/nextpnr
```

mossmann, 2021-02-24:
Additionally I had to `export PYTHONPATH='/usr/local/lib64/trellis'` before `make` would work.

Chiminakust, 2021-05-28:
I've "fixed" this by changing 
`HINTS ${TRELLIS_INSTALL_PREFIX}/lib/${TRELLIS_PROGRAM_PREFIX}trellis`
to
`HINTS ${TRELLIS_INSTALL_PREFIX}/lib64/${TRELLIS_PROGRAM_PREFIX}trellis`
in ecp5/CMakeList.txt line 59.

My output was similar to @mossmann
I hope this can be of use to someone.

closed 2023-09-13 by gatecat