YosysHQ/nextpnr #478: Print better error message when instantiated i2c IP block lacks BUS_ADDR74 defparam

johnMamish, 2020-07-25:
I synthesized the below verilog file (some newlines removed for compactness) with yosys using
`yosys -p "read_verilog i2c_minimal.v; synth_ice40 -top i2c_minimal -json i2c_minimal.json"`

``` Verilog
module i2c_minimal(inout sda, inout scl, input digclk, input sb_wr_i, input sb_stb_i,
	           input [7:0] sb_adr_i, input [7:0] sb_dat_i, output [7:0] sb_dat_o,
	           output sb_ack_o);
    wire sda_in, sda_oe, sda_o, scl_in, scl_oe, scl_o;

    assign sda = sda_oe ? sda_o : 1'bz;
    assign sda_in = sda;
    assign scl = scl_oe ? scl_o : 1'bz;
    assign scl_in = scl;

    SB_I2C i2c1(.SBCLKI(digclk), .SBRWI(sb_wr_i), .SBSTBI(sb_stb_i),
	        .SBADRI7(sb_adr_i[7]), .SBADRI6(sb_adr_i[6]), .SBADRI5(sb_adr_i[5]), .SBADRI4(sb_adr_i[4]),
	        .SBADRI3(sb_adr_i[3]), .SBADRI2(sb_adr_i[2]), .SBADRI1(sb_adr_i[1]), .SBADRI0(sb_adr_i[0]),
	        .SBDATI7(sb_dat_i[7]), .SBDATI6(sb_dat_i[6]), .SBDATI5(sb_dat_i[5]), .SBDATI4(sb_dat_i[4]),
                .SBDATI3(sb_dat_i[3]), .SBDATI2(sb_dat_i[2]), .SBDATI1(sb_dat_i[1]), .SBDATI0(sb_dat_i[0]),
                .SCLI(scl_in), .SDAI(sda_in),
                .SBDATO7(sb_dat_o[7]), .SBDATO6(sb_dat_o[6]), .SBDATO5(sb_dat_o[5]), .SBDATO4(sb_dat_o[4]),
                .SBDATO3(sb_dat_o[3]), .SBDATO2(sb_dat_o[2]), .SBDATO1(sb_dat_o[1]), .SBDATO0(sb_dat_o[0]),
                .SBACKO(sback_o),
                .I2CIRQ(), .I2CWKUP(),
                .SCLO(scl_o), .SCLOE(scl_oe), .SDAO(sda_o), .SDAOE(sda_oe));

    defparam i2c1.I2C_SLAVE_INIT_ADDR = "0b1111100001";
    //defparam i2c1.BUS_ADDR74 = "0b0001";
endmodule
```

and then ran nextpnr-ice40 using
`nextpnr-ice40 --up5k --package sg48 --json i2c_minimal.json --pcf-allow-unconstrained`

Doing this generated an error:
```
terminate called after throwing an instance of 'nextpnr_ice40::assertion_failure'
  what():  Assertion failure: is_string (/home/johnmamish/Downloads/sources/nextpnr/common/nextpnr.h:364)
Aborted (core dumped)
```

johnMamish, 2020-07-25:
I've got a tiny MR that's going up soon which prints
```
ERROR: Invalid value for BUS_ADDR74 for cell i2c1 of type SB_I2C. Valid values include: 0b0001, 0b0011
ERROR: Packing design failed.
```
Instead

gatecat, 2022-08-18:
as a result of some change inbetween a default is now used automatically (afaik, as per the vendor toos) and there's no error on the example anymore