YosysHQ/nextpnr #483: Build failing while looking for pytrellis

jdesiloniz, 2020-07-29:
System: OSX Catalina (mostly fresh install)

While trying to install next-pnr for ECP5 (to use with an ULX3S board) I'm getting stuck at the build phase both while trying to use brew or manual building myself.

On brew I've tried to install it using this tap: https://github.com/ktemkin/homebrew-oss-fpga
`brew install --HEAD project-trellis yosys nextpnr-trellis fujprog`

What I get when it comes to nextpnr-trellis is the following:

```
-- Looking for C++ include stddef.h - found
-- Check size of unsigned short
-- Check size of unsigned short - done
-- Searching 16 bit integer - Using unsigned short
-- Check if the system is big endian - little endian
-- Configuring architecture: ecp5
-- Enabled ECP5 devices: 25k;45k;85k
-- Trellis install prefix: /usr/local
-- Searching for pytrellis in: /Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk/usr/lib;/usr/local/lib;/usr/lib;//lib;/usr/local/Cellar/cmake/3.18.0/lib;/usr/local/Cellar/nextpnr-ecp5/HEAD-b39a2a5/lib;/usr/X11R6/lib;/usr/pkg/lib;/opt/lib;/sw/lib;/opt/local/lib;/usr/lib/X11
CMake Error at ecp5/CMakeLists.txt:64 (message):
  Failed to locate the pytrellis library
```

Similar thing happens when trying to manually build from sources, having built prjtrellis and put it on /opt:
`cmake . -DARCH=ecp5 -DCMAKE_INSTALL_PREFIX=/opt/nextpnr  -DTRELLIS_INSTALL_PREFIX=/opt -DPYTRELLIS_LIBDIR=/opt/trellis/lib/trellis`

Results in the following:

```
-- Found Boost: /usr/local/include (found version "1.72.0") found components: filesystem program_options iostreams system thread regex chrono date_time atomic 
-- Found Boost: /usr/local/include (found version "1.72.0") found components: program_options filesystem system 
-- Configuring architecture: ecp5
-- Enabled ECP5 devices: 25k;45k;85k
-- Trellis install prefix: /opt
-- Searching for pytrellis in: /Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk/usr/lib;/usr/local/lib;/usr/lib;//lib;/usr/local/Cellar/cmake/3.18.0/lib;/opt/nextpnr/lib;/usr/X11R6/lib;/usr/pkg/lib;/opt/lib;/sw/lib;/opt/local/lib;/usr/lib/X11
CMake Error at ecp5/CMakeLists.txt:64 (message):
  Failed to locate the pytrellis library
```

My current setup is mostly fresh as I got this laptop two weeks ago, but I'm surprised I'm not seeing anyone come up with this issue recently. Any clues? Thanks!

smunaut, 2020-07-29:
If the lib is in `/opt/trellis/lib/trellis`, then I would think `-DTRELLIS_INSTALL_PREFIX=/opt/trellis` and not just `/opt`.
Also remove the `-DPYTRELLIS_LIBDIR` all together.

jdesiloniz, 2020-07-29:
@smunaut Thanks for the suggestion! I didn't want to bloat the issue with the multiple attempts I've made so far, but I've tried that too and it results in the same. I've also copied the lib binaries to /opt/lib (as the build script claims trying to find them there) and it didn't have an effect either.

smunaut, 2020-07-29:
Try installing both `trellis` and `nextpnr` in the same prefix.
Just give the same `-DCMAKE_INSTALL_PREFIX=xxx` to both and no other (path related) arguments.

jdesiloniz, 2020-07-29:
@smunaut Thanks again. Your advice makes sense, but unfortunately I'm still seeing the same behavior. I installed trellis on /opt/nextpnr-trellis. Under it I've got:

```
/opt
   /nextpnr-trellis
       /bin
            ecpbram   ecpmulti  ecppack   ecppll    ecpunpack
       /lib
           /trellis
                libtrellis.dylib pytrellis.so
      /share
          /trellis
               /database
               /misc
               /timing
               /util
```

After that, I tried to build nextpnr via the following:

`cmake . -DARCH=ecp5 -DCMAKE_INSTALL_PREFIX=/opt/nextpnr-trellis`

And I got:

```
-- Found Boost: /usr/local/include (found version "1.72.0") found components: filesystem program_options iostreams system thread regex chrono date_time atomic 
-- Found Boost: /usr/local/include (found version "1.72.0") found components: program_options filesystem system 
-- Configuring architecture: ecp5
-- Enabled ECP5 devices: 25k;45k;85k
-- Trellis install prefix: /opt/nextpnr-trellis
-- Searching for pytrellis in: /Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk/usr/lib;/usr/local/lib;/usr/lib;//lib;/usr/local/Cellar/cmake/3.18.0/lib;/opt/nextpnr-trellis/lib;/usr/X11R6/lib;/usr/pkg/lib;/opt/lib;/sw/lib;/opt/local/lib;/usr/lib/X11
CMake Error at ecp5/CMakeLists.txt:64 (message):
  Failed to locate the pytrellis library
```

Based on the logs it seems it's looking at pytrellis at the correct location. Maybe some of the files it needs are missing? I'm not familiar with trellis (as I've used nextpnr only with ice40).

Thanks again :)

smunaut, 2020-07-29:
I don't know much about OSX but the fact it's named `pytrellis.so` and not `pytrellis.dylib` seems weird to me.

jdesiloniz, 2020-07-29:
@smunaut Thanks! That gave me a clue to force cmake to use `pytrellis.so` in the CMakesList.txt file:

```
if(WIN32)
            set(pytrellis_lib pytrellis.pyd)
        else()
            set(pytrellis_lib pytrellis.so)
        endif()
```

And that went through. Unfortunately it fails right away while trying to make the binaries:

```
sudo make
[  1%] Generating chipdb/chipdb-25k.bba
Traceback (most recent call last):
  File "/Users/jdesiloniz/Downloads/nextpnr/ecp5/trellis_import.py", line 21, in <module>
    import pytrellis
ModuleNotFoundError: No module named 'pytrellis'
make[2]: *** [ecp5/chipdb/chipdb-25k.bba] Error 1
make[1]: *** [ecp5/CMakeFiles/chipdb-ecp5-bbas.dir/all] Error 2
make: *** [all] Error 2
```

So I'm not sure if it's not able to correctly link the `pytrellis.so` library at all...

daveshah1, 2020-07-29:
I strongly suspect Python is also looking for a dylib, not a so

jdesiloniz, 2020-07-29:
@daveshah1 that could be the case, or maybe including `.so` library instead of `.dylib` during the cmake phase didn't work even though it didn't throw any errors. The CMakesList.txt file expects the file to be a .dylib (at least that's what happens in Catalina, as `CMAKE_SHARED_LIBRARY_SUFFIX` seems to correspond to `dylib`).

This is the same behavior I see while trying to install through `brew` 😞 . I've tried playing a bit with the dynamic libraries here on my end but unfortunately it doesn't seem to make python accept pytrellis in my local environment.

kost, 2020-10-08:
I think I managed to fix this in latest release in homebrew. 

Please, try to remove project-trellis first:
```
brew remove project-trellis
```

And try to install it again:
```
brew install --HEAD project-trellis yosys nextpnr-trellis fujprog
```

Let me know if it works for you.

sthornington, 2020-10-26:
I am having the same issue.

wizz777, 2020-11-17:
I am too:
-- Found Boost: /usr/local/include (found version "1.74.0") found components: program_options filesystem system
-- Configuring architecture: generic
-- Configuring architecture: ice40
-- Enabled iCE40 devices: 384;1k;5k;u4k;8k
-- IceStorm install prefix: /usr/local
-- icebox data directory: /usr/local/share/icebox
-- Using iCE40 chipdb: /Users/wizz/Downloads/nextpnr-master/build/ice40/chipdb
-- Configuring architecture: ecp5
-- Enabled ECP5 devices: 25k;45k;85k
-- Trellis install prefix: /usr/local
-- Searching for pytrellis in: /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX11.0.sdk/usr/lib;/usr/local/lib;/usr/lib;//lib;/usr/local/Cellar/cmake/3.18.4/lib;/usr/local/lib;/usr/X11R6/lib;/usr/pkg/lib;/opt/lib;/sw/lib;/opt/local/lib;/usr/lib/X11
CMake Error at ecp5/CMakeLists.txt:64 (message):
  Failed to locate the pytrellis library

sthornington, 2020-11-17:
For people who just want to get to programming boards, I had success using the binaries built here: https://github.com/open-tool-forge/fpga-toolchain

wizz777, 2020-11-17:
> For people who just want to get to programming boards, I had success using the binaries built here: https://github.com/open-tool-forge/fpga-toolchain

Thank you!

davidcorrigan714, 2021-01-03:
I'm seeing this on a windows build too, really seems like the install script in prjtrellis isn't bundling pytrellis properly.

olegendo, 2021-07-02:
Similar issue on Fedora:

```

- Searching for pytrellis in: /usr/local/lib;/usr/lib;//lib;/usr/lib;/usr/local/lib;/usr/X11R6/lib;/usr/pkg/lib;/opt/lib;/usr/lib/X11
CMake Error at ecp5/CMakeLists.txt:64 (message):
  Failed to locate the pytrellis library

-- Configuring incomplete, errors occurred!

```

The following fixes the cmake error for me

```
diff --git a/ecp5/CMakeLists.txt b/ecp5/CMakeLists.txt
index bc34d51b..599d9afd 100644
--- a/ecp5/CMakeLists.txt
+++ b/ecp5/CMakeLists.txt
@@ -49,6 +49,8 @@ else()
         endif()
         message(STATUS "Searching for pytrellis in: ${pytrellis_paths}")
 
+        list(APPEND pytrellis_paths "/usr/lib64")
+
         if(WIN32)
             set(pytrellis_lib pytrellis.pyd)
         else()
diff --git a/tests b/tests
index ccc61e5e..8f93e7e0 160000
--- a/tests
+++ b/tests
@@ -1 +1 @@

```

OpenEDF, 2021-10-16:
I am having the same issue.
-- Building with IPO
-- Found Boost: /usr/include (found version "1.71.0") found components: filesystem program_options iostreams system thread regex chrono date_time atomic
-- Found Boost: /usr/include (found version "1.71.0") found components: program_options filesystem system
-- Configuring architecture: ecp5
-- Enabled ECP5 devices: 25k;45k;85k
-- Trellis install prefix: /usr/localsudo
-- Searching for pytrellis in: /usr/local/lib;/usr/local/lib/x86_64-linux-gnu;/usr/lib;/usr/lib/x86_64-linux-gnu;//lib;//lib/x86_64-linux-gnu;/usr/lib;/usr/lib/x86_64-linux-gnu;/usr/local/lib;/usr/local/lib/x86_64-linux-gnu;/usr/X11R6/lib;/usr/X11R6/lib/x86_64-linux-gnu;/usr/pkg/lib;/usr/pkg/lib/x86_64-linux-gnu;/opt/lib;/opt/lib/x86_64-linux-gnu;/usr/lib/X11
CMake Error at ecp5/CMakeLists.txt:64 (message):
  Failed to locate the pytrellis library

olofk, 2023-05-17:
Same issue here. Trellis installed into `$HOME/.local`. I got around it by manually adding `list(APPEND pytrellis_paths "/home/olof/.local/lib64")` to CMakeLists as was suggested above. It looks like the CMake script isn't looking for pytrellis in the dir where I told it trellis is installed

closed 2024-09-26 by gatecat