YosysHQ/nextpnr #495: "Assertion failure: closest != BelId() on .../nextpnr/ecp5/pack.cc:2610

aventuri, 2020-09-02:
i was just testing compiling a litex  gateware for "logicbone" and got this assertion error on nextpnr:

====
....
4.52. Executing JSON backend.

Warnings: 1 unique messages, 1 total
End of script. Logfile hash: fcee3e6299, CPU: user 41.40s system 0.69s, MEM: 1475.03 MB peak
Yosys 0.9+3556 (git sha1 3f27a4ea6, clang 7.0.1-8 -fPIC -Os)
Time spent: 19% 8x techmap (9 sec), 13% 33x opt_clean (6 sec), ...
Info: constraining clock net 'refclk' to 25.00 MHz

Info: Logic utilisation before packing:
Info:     Total LUT4s:      8265/43848    18%
Info:         logic LUTs:   7105/43848    16%
Info:         carry LUTs:    656/43848     1%
Info:           RAM LUTs:    336/21924     1%
Info:          RAMW LUTs:    168/10962     1%

Info:      Total DFFs:      7066/43848    16%

Info: Packing IOs..
Info: pin 'ddram_cas_n$tr_io' constrained to Bel 'X0/Y65/PIOB'.
Info: pin 'ddram_cke$tr_io' constrained to Bel 'X0/Y44/PIOA'.
....
Info: IOLOGIC component ODDRX2F_8 connected to PIO Bel X0/Y11/PIOB
Info: IOLOGIC component ODDRX2F_9 connected to PIO Bel X0/Y14/PIOC
terminate called after throwing an instance of 'nextpnr_ecp5::assertion_failure'
  what():  **Assertion failure: closest != BelId() (/home/andrea/src/fpga/tools/nextpnr/ecp5/pack.cc:2610)**
Aborted
andrea@av-ideal:~/src/fpga/tools/litex/litex-boards/litex_boards/targets/build/logicbone/gateware$ ls
build_logicbone.sh  logicbone.json  logicbone.lpf  logicbone.rpt  logicbone.v  logicbone.ys  mem_1.init  mem_2.init  mem.init
====

this output comes more or less a fresh checkout of every tool involved as of today..
**nextpnr** commit 4512a9d
**litex-boards** commit 9b6ed6

i'm stopping here on report as i have no more clues.. of course i'm available for providing other info on request

daveshah1, 2020-09-05:
I can't reproduce this here. Please can you provide the design files and the commands you ran to generate them?

Also, have you checked that your prjtrellis database is up to date (if it isn't then update it, re-install libtrellis, run `touch ecp5/trellis_import.py` in nextpnr and then rebuild nextpnr)?

aventuri, 2020-09-07:
hi Dave, i tried as you say with thw "touch ..." but then, after a nextpnr rebuild&install the output was the same error for the "logicbone" design.
currently the "design files" are the "vanilla" src codes i've got from https://github.com/litex-hub/litex-boards with last commit 9b6ed6bd
BTW i tested to build some board gatewares to grasp the concepts and versa_ecp5, icebreaker, ulx3s were fine; the logicbone was (and still is) throwing out that "assert"
i'l try other boards tests in the evening, eventually

closed 2022-08-18 by gatecat