YosysHQ/nextpnr #497: ice40up5k: cell type 'SB_IO_OD' is unsupported

cwolpert, 2020-09-09:
Hi,
I want to configure ice40UP5K's RGB pins as user IOs, similar to what is described in Annex B in TN1288 "iCE40 LED Driver Usage Guide" from lattice and the test code I found on https://github.com/YosysHQ/icestorm/blob/master/icefuzz/tests/sb_io_od.v. This is my module:
```
// configure ice40UP RGB IOs as GPIOs
// iCE40 LED Driver Usage Guide
module rgb_as_gpio(input a, output o1);
	wire o1i;
	
	assign o1i = a;

	SB_IO_OD #(
		.PIN_TYPE(6'b010001),
		.NEG_TRIGGER(1'b0)
	) IO_PIN_I (
		.PACKAGEPIN(o1),
		.LATCHINPUTVALUE(),
		.CLOCKENABLE(),
		.INPUTCLK(),
		.OUTPUTCLK(),
		.OUTPUTENABLE(),
		.DOUT0(o1i),
		.DOUT1(),
		.DIN0(),
		.DIN1()
	);
endmodule
```
I instantiate it with "rgb_as_gpio RGB0 (.a(micout_r), .o1(AUD_R));", however nextpnr complains: 

ERROR: cell type 'SB_IO_OD' is unsupported (instantiated as 'RGB0.IO_PIN_I')

I got the same error with lattice's original code from the above mentioned TN.

Could someone help me please?

daveshah1, 2020-09-10:
For now just use the RGB pins like any other IO as a workaround (I will add SB_IO_OD to my todo list, it was supported by arachne but never moved across to nextpnr)

pablogs9, 2020-12-12:
Same issue here, I have a kind of offtopic question. I forget to add the pull-up resistors so these RGB open-drain pins are not working in my design (I'm using them as GPIO). Is there any chance of enabling an internal pull-up?

Thanks!

cwolpert, 2021-01-01:
Hi pablogs9,
according to TN1288 there are no pull-ups for those RGP GP(I)Os. I forgot them as well and had to rework my PCB. But now these outputs are working fine. I'm using one the 3 pins as input, that one is working well, too.
Cheers,
Chris

thesourcerer8, 2025-01-30:
In case anyone needs SB_IO_OD: I was able to replace it with SB_IO instead, I only had to rename the parameters slightly.