YosysHQ/nextpnr #510: Design of a Gowin target

pepijndevos, 2020-10-26:
I spent a bunch of time reading the documentation and code over the past few days, so I'm getting ready to start writing code, but I also have a lot of questions. So rather than a constant stream of questions on IRC, I collected my questions to do it a bit more organised.

# Step 0: get Apicula ready

My first concern is translating the Python chipdb to a bba structure that is suitable for nextpnr. Here is the current structure: https://github.com/YosysHQ/apicula/blob/de4cb3b801a9eee4e5b1486aa5690b840b9f2285/chipdb.py

Basically a Device has a grid of (deduplicated) Tiles, and a dict of aliases for global wires. Each Tile has a dict of bels and pips indexed by bel type or source/destination wire. These are local wire names that are mapped to global wires by the `wire2global` function. At the leafs of these dicts are sets of bit locations, which nextpnr doesn't need.

Of course that all has to become constids, arrays, and structs. And I guess since it already works with nextpnr-generic not too many changes would be needed. But I have two concerns.

1. How does lookup work? Do you rebuild a C++ hashmap at the other side, sort them and binary search, or something else?
2. Since aliases turn out to be not supported, should this be revisited? I don't fully understand your neighbourhood concept.

Clock routing on Gowin is connected kind of arbitrarily, so I'm not sure what a good alternative is that will not blow up the amount of tile types. Right now they are all the same type, but global aliases connect the various clock pips between tiles.

[image]

# Step 1: define data types and headers

In `archdefs.h` a bunch of types are defined. First of all the delay types. I'm thinking of going with float, since that's what the vendor db provides, does this come at a speed cost, or it's negligible? A more puzzling thing to match up is the DelayInfo.

My current understanding is that Gowin provides 4 timing values per "thing", for all combinations of input/output rising/falling of a given "thing". https://github.com/YosysHQ/apicula/blob/master/doc/filestructure.md#timing-file It appears nextpnr provides min/max rise/fall delay. What would be the correct mapping here?

Then there are a bunch of `*Id` which the docs say are just representations of the name of an item. But then goes on to say they need to implement comparison operators. Currently my wires have names like `R5C6_N280`, but it seems odd to say this is greater or smaller than `R5C7_N240`. The nexus target seems to use a linear `tile` and `index` value, of which the semantics are not yet clear to me. I'm not even 101% sure if these are global names like `R5C7_N240` or just map directly to my chipdb, which is a tile-local `N240`, but I assume they should be global.

# Step 2: draw the rest of the owl

This is of course the biggest work, but also seems the most straightforward in a way. There are a lot of methods, but a lot of them have trivial implementations.

Some things I'm kinda unsure about:
1. Custom packing steps. I probably need to pack a LUT+DFF into a slice. Or maybe even two, because they share a clock and reset, but IIRC it is preferred to use constraints for that. I can probably take inspiration from various places for this.
2. Constraints. I don't really do this with nextpnr-generic. Probably not so hard.
3. FASM output? Currently just using JSON directly, which works fine. Only really an issue if there were ever to exist another PnR implementation for Gowin.
3. Testing along the way. Do I have to implement the whole API and then wade through the errors, or can I implement smaller chunks and test them in isolation?

daveshah1, 2020-10-26:
>     1. How does lookup work? Do you rebuild a C++ hashmap at the other side, sort them and binary search, or something else?

Lookups are generally done by flat indices for anything important. The one time where hashmap or searches are useful is the mapping from name to object (which is rarely if ever used).

>     2. Since aliases turn out to be not supported, should this be revisited? I don't fully understand your neighbourhood concept.

To explain neighbourhoods:

 - Take all the wires in a tile
 - For each wire in the tile; look at the other tiles it is connected to (using relative coordinates*) and the wire name in that tile
 - The combination of those is a neighbourhood

The unique kinds of neighbourhood are then deduplicated.

* with some special cases for global wires to improve deduplication quality

> [image]
> # Step 1: define data types and headers
>
> In `archdefs.h` a bunch of types are defined. First of all the delay types. I'm thinking of going with float, since that's what the vendor db [...]

The speed cost will be negligible. 

> My current understanding is that Gowin provides 4 timing values per "thing", for all combinations of input/output rising/falling of a given "thing". [...]

Nothing in nextpnr cares about unateness at present, so doesn't really matter (probably map {falling, falling} to falling and {rising, rising} to rising right now). This might need to be a future API change to support non-unate delays.

> Then there are a bunch of `*Id` which the docs say are just representations of the name of an item. But then goes on to say they need to implement comparison operators. Currently my wires have names [...]

These should be flat indices for best performance. In a deduplicated database they would generally have two fields; the location of a base tile and the index within that tile.

> # Step 2: draw the rest of the owl
>
> This is of course the biggest work, but also seems the most straightforward in a way. There are a lot of methods, but a lot of them have trivial implementations. [...]

Assuming that LUT and DFF can never be used independently, then one LUT+DFF is a correct unit.

>     2. Constraints. I don't really do this with nextpnr-generic. Probably not so hard.

Have a look at arch_place.cc in nexus - this is slightly more complex than needed due to incremental updates but should give you a good idea what's needed.

>     3. FASM output? Currently just using JSON directly, which works fine. Only really an issue if there were ever to exist another PnR implementation for Gowin.

Your choice

>     4. Testing along the way. Do I have to implement the whole API and then wade through the errors, or can I implement smaller chunks and test them in isolation?

You can get bel stuff done and test placement works; then implement wire/pip stuff and make routing work. The Python API is very useful for debugging what your Arch is doing.

TheFlav, 2021-06-23:
I have no idea if this is the right way to ask for help with this.  I'm using the Tang Nano board to learn about Gowin FPGAs under Mac.  I'd prefer to not need Windows, so I'm trying to use the open-source toolchain.

I want to try to build https://github.com/dotcypress/tang-nano-lcd using yosys, nextpnr-gowin, etc.

When I do so, nextpnr-gowin errors out with this message.
ERROR: Unable to place cell 'rpll.rpll_inst', no BELs remaining to implement cell type 'rPLL'

I don't even know where to begin looking for a solution to this.

pepijndevos, 2021-06-24:
Apicula does not support PLL yet.
PLL and BRAM are my top priorities for Apicula, but Apicula is not my top priority right now.

roman3017, 2022-04-11:
Is this also a known issue when building blinky-osc-tangnano9k.fs with nextpnr-gowin?
```
ERROR: Unable to place cell 'osc', no BELs remaining to implement cell type 'OSC'
0 warnings, 1 error
make: *** [Makefile:79: blinky-osc-tangnano9k.json] Error 255
```

pepijndevos, 2022-04-11:
Is your nextpnr up to date? This was only recently added.

roman3017, 2022-04-12:
I see. I rebuilt nextpnr yesterday from the latest. But I am using apicula from the official pip install, since "make" does not compile for me with Gowin software 1.9.7. And I cannot find 1.9.8 for Linux. I guess I will need to wait for the next release of apicula then. Thanks

pepijndevos, 2022-04-12:
Ah great. I made a new release anyway: https://github.com/YosysHQ/apicula/releases/tag/0.3.1

closed 2023-09-13 by gatecat