YosysHQ/nextpnr #517: MacOS - failed to locate pytrellis library

BracketMaster, 2020-11-17:
With Cmake 3.19 which is the default cmake that comes with 
brew, building nextpnr on Mac fails. This is because CMake searches for
``pytrellis.dylib``, but prjtrellis build ``pytrellis.so``.

whitequark, 2020-11-17:
That seems like something to be fixed in prjtrellis, not nextpnr.

BracketMaster, 2020-11-17:
ok. fair.

BracketMaster, 2020-11-17:
it seems like you can specify SHARED_LIBRARY
or MODULE in CMAKE which are interpreted to mean
.dylib or .so respectively in MacOS.

I'm wondering whether this is something that prjtrellis
should change, or NextPNR should change?

BracketMaster, 2020-11-17:
This line is the culprit:

https://github.com/YosysHQ/nextpnr/blob/master/ecp5/CMakeLists.txt#L55

Doing the following seems to work for all platforms?

```
        if(WIN32)
            set(pytrellis_lib pytrellis.pyd)
        elif(APPLE)
            set(pytrellis_lib pytrellis.so)
        else()
            set(pytrellis_lib pytrellis${CMAKE_SHARED_LIBRARY_SUFFIX})
        endif()
```

whitequark, 2020-11-17:
Isn't this the opposite of what you described in the issue?

BracketMaster, 2020-11-17:
Ok. You're right.
I forgot to clean some cruft and CMAKE wasn't 
actually doing anything useful.

It should be this.
```
        if(WIN32)
            set(pytrellis_lib pytrellis.pyd)
        elseif (APPLE)
            set(pytrellis_lib pytrellis.so)
        else()
            set(pytrellis_lib pytrellis${CMAKE_SHARED_LIBRARY_SUFFIX})
            message("hello ${CMAKE_SYSTEM_NAME}")
        endif()
```

BracketMaster, 2020-11-17:
The following is notable cleaner and works on both MacOS
and Linux. I think Prjtrellis has CMAKE treat pytrellis as a shared
module, not a shared library.

@whitequark, do you think this would make sense to open a pull request
for review?
```
        if(WIN32)
            set(pytrellis_lib pytrellis.pyd)
        else()
            set(pytrellis_lib pytrellis${CMAKE_SHARED_MODULE_SUFFIX})
            message("hello ${CMAKE_SYSTEM_NAME}")
        endif()
```

whitequark, 2020-11-17:
LGTM

BracketMaster, 2020-11-17:
I actually remember hitting this issue last winter on the previous
version of MacOS. Does NextPNR have a MacOS CI?

I hardly update NextPNR until I update MacOS - the ARM
Macs just came out

closed 2020-11-18 by BracketMaster