YosysHQ/nextpnr #521: ChipDBs build flag for windows

davidcorrigan714, 2020-11-28:
I'm working on getting nextpnr & prjtrellis compiled on windows and I've made good progress except something seems off about loading the chipdbs - namely that it seems to take an extra compiler flag to make it work properly on windows. Browsing through the code I see it's looking for the WIN32 flag, which seems to pick the right loading function if I add it to the compiler command line, but maybe it should be _WIN32 to pick up windows builds automatically?

whitequark, 2020-11-28:
The situation with `WIN32` and `_WIN32` is a bit [complicated](https://stackoverflow.com/a/662543/254415); in short, `_WIN32` is always defined by a compiler targeting Windows, and `WIN32` is defined somewhere else. That "somewhere else" is the project template in case of MSVC, and the system headers in case of MinGW.

Your conclusion is correct, `common/embed.{cc,h}` should use `_WIN32` instead of `WIN32`.

davidcorrigan714, 2020-11-28:
I'll submit a PR for it in the next few days.