YosysHQ/nextpnr #523: nextpnr will not compile on new Apple Powerbook M1

bmentink, 2020-11-29:
Hi all,
I have compiled the icetools and yosys fine on this machine but nextpnr does not build.
I am building in a Rosseta 2 emulation terminal session, so it should be building for x86, but it seems to pick up that it is building for the host ARM64.

It fails in the "bba" module, the error is:

`
[ 50%] Linking CXX executable bbasm
ld: warning: ignoring file /usr/local/lib/libboost_program_options-mt.dylib, building for macOS-arm64 but attempting to link with file built for macOS-x86_64
ld: warning: ignoring file /usr/local/lib/libboost_filesystem-mt.dylib, building for macOS-arm64 but attempting to link with file built for macOS-x86_64
ld: warning: ignoring file /usr/local/lib/libboost_system-mt.dylib, building for macOS-arm64 but attempting to link with file built for macOS-x86_64
Undefined symbols for architecture arm64:
  "boost::program_options::to_internal(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&)", referenced from:
      std::__1::vector<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::allocator<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > > > boost::program_options::to_internal<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > >(std::__1::vector<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::allocator<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > > > const&) in main.cc.o
  "boost::program_options::variables_map::variables_map()", referenced from:
      _main in main.cc.o
  "boost::program_options::options_description::add_options()", referenced from:
      _main in main.cc.o
  "boost::program_options::options_description::m_default_line_length", referenced from:
      _main in main.cc.o
  "boost::program_options::options_description::options_description(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, unsigned int, unsigned int)", referenced from:
      _main in main.cc.o
  "boost::program_options::invalid_option_value::invalid_option_value(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&)", referenced from:
      void boost::program_options::validate<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, char>(boost::any&, std::__1::vector<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::allocator<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > > > const&, std::__1::vector<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::allocator<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > > >*, int) in main.cc.o
  "boost::program_options::options_description_easy_init::operator()(char const*, boost::program_options::value_semantic const*, char const*)", referenced from:
      _main in main.cc.o
  "boost::program_options::options_description_easy_init::operator()(char const*, char const*)", referenced from:
      _main in main.cc.o
  "boost::program_options::positional_options_description::add(char const*, int)", referenced from:
      _main in main.cc.o
  "boost::program_options::positional_options_description::positional_options_description()", referenced from:
      _main in main.cc.o
  "boost::program_options::arg", referenced from:
      boost::program_options::typed_value<std::__1::vector<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::allocator<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > > >, char>::name() const in main.cc.o
  "boost::program_options::store(boost::program_options::basic_parsed_options<char> const&, boost::program_options::variables_map&, bool)", referenced from:
      _main in main.cc.o
  "boost::program_options::detail::cmdline::set_positional_options(boost::program_options::positional_options_description const&)", referenced from:
      _main in main.cc.o
  "boost::program_options::detail::cmdline::set_options_description(boost::program_options::options_description const&)", referenced from:
      _main in main.cc.o
  "boost::program_options::detail::cmdline::get_canonical_option_prefix()", referenced from:
      boost::program_options::basic_command_line_parser<char>::run() in main.cc.o
  "boost::program_options::detail::cmdline::run()", referenced from:
      boost::program_options::basic_command_line_parser<char>::run() in main.cc.o
  "boost::program_options::detail::cmdline::cmdline(std::__1::vector<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::allocator<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > > > const&)", referenced from:
      boost::program_options::basic_command_line_parser<char>::basic_command_line_parser(int, char const* const*) in main.cc.o
  "boost::program_options::notify(boost::program_options::variables_map&)", referenced from:
      _main in main.cc.o
  "boost::program_options::validate(boost::any&, std::__1::vector<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::allocator<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > > > const&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >*, int)", referenced from:
      void boost::program_options::validate<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, char>(boost::any&, std::__1::vector<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::allocator<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > > > const&, std::__1::vector<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::allocator<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > > >*, int) in main.cc.o
  "boost::program_options::operator<<(std::__1::basic_ostream<char, std::__1::char_traits<char> >&, boost::program_options::options_description const&)", referenced from:
      _main in main.cc.o
  "boost::filesystem::path::stem() const", referenced from:
      _main in main.cc.o
  "boost::program_options::abstract_variables_map::operator[](std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&) const", referenced from:
      _main in main.cc.o
  "boost::program_options::error_with_option_name::substitute_placeholders(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&) const", referenced from:
      vtable for boost::wrapexcept<boost::program_options::invalid_option_value> in main.cc.o
      vtable for boost::program_options::invalid_option_value in main.cc.o
  "boost::program_options::error_with_option_name::what() const", referenced from:
      vtable for boost::wrapexcept<boost::program_options::invalid_option_value> in main.cc.o
      vtable for boost::program_options::invalid_option_value in main.cc.o
  "boost::program_options::value_semantic_codecvt_helper<char>::parse(boost::any&, std::__1::vector<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::allocator<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > > > const&, bool) const", referenced from:
      vtable for boost::program_options::typed_value<std::__1::vector<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::allocator<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > > >, char> in main.cc.o
  "typeinfo for boost::program_options::error_with_option_name", referenced from:
      typeinfo for boost::program_options::validation_error in main.cc.o
  "typeinfo for boost::program_options::value_semantic_codecvt_helper<char>", referenced from:
      typeinfo for boost::program_options::typed_value<std::__1::vector<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::allocator<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > > >, char> in main.cc.o
[... cut ...]
      boost::wrapexcept<boost::program_options::invalid_option_value>::~wrapexcept() in main.cc.o
      non-virtual thunk to boost::wrapexcept<boost::program_options::invalid_option_value>::~wrapexcept() in main.cc.o
      ...
  NOTE: a missing vtable usually means the first non-inline virtual member function has no definition.
ld: symbol(s) not found for architecture arm64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [bba/bbasm] Error 1
make[1]: *** [bba/CMakeFiles/bbasm.dir/all] Error 2
make: *** [all] Error 2
`

daveshah1, 2020-11-29:
@ktemkin - you had the tools working on a new Mac - would you be able to help here?

bmentink, 2020-11-29:
I will help where I can, will need guidance though :)

ktemkin, 2020-11-30:
> @ktemkin - you had the tools working on a new Mac - would you be able to help here?

I do, but I'm building them natively, rather than through the Rosetta2 translation layer. If anyone's interested in doing this, building using ktemkin/homebrew-oss-fpga and a bunch of work; some of my hacks are under ktemkin/homebrew-tirepatch.

If I remember, I'll try this via Rosetta2 and see if I can't spot if there are any issues that need to be fixed on the nextpnr side.

bmentink, 2020-11-30:
@ktemkin 
Native is good. ! Will your changes go into the main branch anytime, or should I build your branch to try ..

bmentink, 2020-11-30:
I tried brew'ing from your repo, but I get the following (in a normal terminal, not Rosetta2 terminal):

`
Bernards-Air:~ bmentink$ brew install --HEAD nextpnr-ice40
Updating Homebrew...
Error: Cannot install in Homebrew on ARM processor in Intel default prefix (/usr/local)!
Please create a new installation in /opt/homebrew using one of the
"Alternative Installs" from:
  https://docs.brew.sh/Installation
You can migrate your previously installed formula list with:
  brew bundle dump
`

Should I be brew'ing in a Rosetta2 terminal or normal ARM terminal?
I am a bit confused here ..

aep, 2020-12-02:
fwiw, the  part that fails for you compiles fine on native darwin-arm64. 

unfortunately Qt5 wont compile because libuci doesnt support arm, so we'll have to wait for that

whitequark, 2020-12-02:
Are you aware that Qt5 is an optional dependency?

aep, 2020-12-02:
ah, i wasnt, thanks.

compiles fine for me on native arm64. binary attached for analysis although its not very useful without its dependencies.

i had to manually patch bba/CMakeLists.txt tho to add boost include path because i dont know cmake works.

also PR for minor fix in https://github.com/YosysHQ/nextpnr/pull/529

bmentink, 2020-12-03:
Can you please tell me how to compile for arm64? I don't know how to do that, as my home-brew is only x86.

aep, 2020-12-03:
I just use macports. it's far less complex and supports multi arch out of the box.

bmentink, 2020-12-03:
Ahhh, I see you just have to do: export ARCHFLAGS='-arch arm64'. .. before the brew command ... :)

bmentink, 2020-12-09:
Sill having issues, anyone have a process of building the tools using brew .. please indicate the steps ..

bmentink, 2020-12-10:
@aep 
Since I can't get home-brew working, can you give me the steps on installing the tools (icetools,yosys,nextpnr) using macports.

Many Thanks.

aep, 2020-12-11:
@bmentink i just discoverd there's actually official installation instructions for yosys on macports here:

https://github.com/YosysHQ/yosys#setup

but what i did:

1. see installation instructions at https://www.macports.org/install.php
2. sudo port install cmake pytthon38 boost eigen3  # and maybe other stuff?
3. patch CMakeFiles manually to add the includepaths
4. patch icestorm to remove GNU dependencies. https://github.com/YosysHQ/icestorm/pull/274
5. disable Qt gui with ccmake, because i couldnt be bothered getting Qt working

it's pretty smooth from there.

ktemkin, 2020-12-11:
> Sill having issues, anyone have a process of building the tools using brew .. please indicate the steps ..

I replied to your issue related to this over in https://github.com/ktemkin/homebrew-oss-fpga/issues/1.

bmentink, 2020-12-11:
@aep 
Thanks. I have installed MacPorts and done step 2, but am missing a couple of steps before 3.
How do I download the source using  port, so I can modify the files you mention ...

@ktemkin 
Thanks also, I will try your modified gcc (or the one produced by port) :) .. and try again to brew nextpnr

aep, 2020-12-14:
git comes with xcode, you just git clone this repo.

gatecat, 2022-08-18:
most of the issues here should be solved; for anyone visiting this in the future prebuilt arm64 osx builds should also be available from https://github.com/YosysHQ/oss-cad-suite-build/releases

closed 2022-08-18 by gatecat