YosysHQ/nextpnr #531: Virtualbox Error in ubuntu 18.04

Dhyansaj, 2020-12-04:
Its great annoying to hear this. I have reinstalled ubuntu 18.04 and started installing apps that I need. Downloaded q4wine. But didnt open. When starting from terminel it says no file.......bla bla. Thought of installing virtualbox. The same error repeated.

[image]

Dhyansaj, 2020-12-04:
Could you please solve this

daveshah1, 2020-12-04:
I'm not sure I understand what the link with nextpnr is?

Dhyansaj, 2020-12-04:
Can you try ?  Or will the issue be solved if i reinstalled ubuntu ?

daveshah1, 2020-12-04:
I am trying to understand why you have filed this as an issue against nextpnr? Did the problem start when you installed nextpnr?  I don't use Ubuntu myself, and this isn't the right forum for general Ubuntu/Qt questions in any case.

Xiretza, 2020-12-04:
https://github.com/YosysHQ/nextpnr/issues/375 is the second google result for `libQt5Core.so.5: cannot open shared object file: No such file or directory`, so they may have just blindly posted their problem into the first issue tracker they found.

closed 2020-12-04 by Dhyansaj

daveshah1, 2020-12-04:
@Xiretza I think you must be right. I guess it's free SEO though :D

[image]