YosysHQ/nextpnr #532: build error

arblake, 2020-12-04:
Building get to this point:
[ 51%] Building CXX object CMakeFiles/nextpnr-ecp5.dir/ecp5/pack.cc.o
/home/adrian/nextpnr/ecp5/baseconfigs.cc: In function ‘void nextpnr_ecp5::BaseConfigs::config_empty_lfe5u_45f(nextpnr_ecp5::ChipConfig&)’:
/home/adrian/nextpnr/ecp5/baseconfigs.cc:190:6: note: variable tracking size limit exceeded with ‘-fvar-tracking-assignments’, retrying without
  190 | void config_empty_lfe5u_45f(ChipConfig &cc)
      |      ^~~~~~~~~~~~~~~~~~~~~~
[ 51%] Building CXX object CMakeFiles/nextpnr-ecp5.dir/ecp5/pio.cc.o
/home/adrian/nextpnr/ecp5/bitstream.cc: In function ‘void nextpnr_ecp5::write_bitstream(nextpnr_ecp5::Context*, std::string, std::string)’:
/home/adrian/nextpnr/ecp5/bitstream.cc:578:6: note: variable tracking size limit exceeded with ‘-fvar-tracking-assignments’, retrying without
  578 | void write_bitstream(Context *ctx, std::string base_config_file, std::string text_config_file)
      |      ^~~~~~~~~~~~~~~
/home/adrian/nextpnr/ecp5/baseconfigs.cc: In function ‘void nextpnr_ecp5::BaseConfigs::config_empty_lfe5u_85f(nextpnr_ecp5::ChipConfig&)’:
/home/adrian/nextpnr/ecp5/baseconfigs.cc:523:6: note: variable tracking size limit exceeded with ‘-fvar-tracking-assignments’, retrying without
  523 | void config_empty_lfe5u_85f(ChipConfig &cc)
      |      ^~~~~~~~~~~~~~~~~~~~~~
{standard input}: Assembler messages:
{standard input}:27314269: Warning: end of file not at end of a line; newline inserted
{standard input}:27314300: Error: unknown pseudo-op: `.qua'
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
make[2]: *** [CMakeFiles/nextpnr-ecp5.dir/build.make:375: CMakeFiles/nextpnr-ecp5.dir/ecp5/baseconfigs.cc.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:265: CMakeFiles/nextpnr-ecp5.dir/all] Error 2
make: *** [Makefile:130: all] Error 2

Building using latest Ubuntu

daveshah1, 2020-12-04:
Looks like the cpmpiler is running out of memory. Can you try building with fewer parallel jobs?

arblake, 2020-12-04:
I did  make -j2 and still the same error.

daveshah1, 2020-12-05:
How much free memory do you have before starting the build?

arblake, 2020-12-05:
I think that was the problem . Needs almost 8G RAM
Look:
[ 51%] Building CXX object CMakeFiles/nextpnr-ecp5.dir/ecp5/baseconfigs.cc.o
/home/adrian/nextpnr/ecp5/baseconfigs.cc: In function ‘void nextpnr_ecp5::BaseConfigs::config_empty_lfe5u_45f(nextpnr_ecp5::ChipConfig&)’:
/home/adrian/nextpnr/ecp5/baseconfigs.cc:190:6: note: variable tracking size limit exceeded with ‘-fvar-tracking-assignments’, retrying without
  190 | void config_empty_lfe5u_45f(ChipConfig &cc)
      |      ^~~~~~~~~~~~~~~~~~~~~~
/home/adrian/nextpnr/ecp5/baseconfigs.cc: In function ‘void nextpnr_ecp5::BaseConfigs::config_empty_lfe5u_85f(nextpnr_ecp5::ChipConfig&)’:
/home/adrian/nextpnr/ecp5/baseconfigs.cc:523:6: note: variable tracking size limit exceeded with ‘-fvar-tracking-assignments’, retrying without
  523 | void config_empty_lfe5u_85f(ChipConfig &cc)
      |      ^~~~~~~~~~~~~~~~~~~~~~
[ 51%] Linking CXX executable nextpnr-ecp5
[ 51%] Built target nextpnr-ecp5
adrian@adrian-HP-Spectre-Notebook:~/nextpnr$ 

no errors

closed 2020-12-27 by daveshah1