YosysHQ/nextpnr #534: fatal error: Python.h: No such file or directory

gojimmypi, 2020-12-06:
TL;DR:  cmake cache!

I was having a problem building `nextpnr-ice40` on WSL as [discussed on Twitter with @BrunoLevy](https://twitter.com/gojimmypi/status/1335687264362369024?s=20). The issue was I kept seeing this build problem about `fatal error: Python.h: No such file or directory`:

```
$ make -j$(nproc)
[  3%] Automatic MOC for target gui_ice40
[  7%] Automatic MOC for target QtPropertyBrowser
[  9%] Built target chipdb-ice40-bbas
[  9%] Built target bbasm
[  9%] Built target gui_ice40_autogen
[  9%] Built target QtPropertyBrowser_autogen
[ 14%] Built target chipdb-ice40-bins
[ 24%] Built target chipdb-ice40
[ 36%] Built target QtPropertyBrowser
[ 37%] Building CXX object generated/gui/ice40/CMakeFiles/gui_ice40.dir/pythontab.cc.o
[ 38%] Building CXX object generated/gui/ice40/CMakeFiles/gui_ice40.dir/__/3rdparty/python-console/modified/pyredirector.cc.o
[ 39%] Building CXX object generated/gui/ice40/CMakeFiles/gui_ice40.dir/__/3rdparty/python-console/modified/pyinterpreter.cc.o
In file included from /mnt/c/workspace/nextpnr/3rdparty/python-console/modified/pyredirector.cc:24:
/mnt/c/workspace/nextpnr/3rdparty/python-console/modified/pyredirector.h:26:10: fatal error: Python.h: No such file or directory
   26 | #include <Python.h>
      |          ^~~~~~~~~~
compilation terminated.
make[2]: *** [generated/gui/ice40/CMakeFiles/gui_ice40.dir/build.make:340: generated/gui/ice40/CMakeFiles/gui_ice40.dir/__/3rdparty/python-console/modified/pyredirector.cc.o] Error 1
make[2]: *** Waiting for unfinished jobs....
/mnt/c/workspace/nextpnr/3rdparty/python-console/modified/pyinterpreter.cc:25:10: fatal error: Python.h: No such file or directory
   25 | #include <Python.h>
      |          ^~~~~~~~~~
compilation terminated.
make[2]: *** [generated/gui/ice40/CMakeFiles/gui_ice40.dir/build.make:353: generated/gui/ice40/CMakeFiles/gui_ice40.dir/__/3rdparty/python-console/modified/pyinterpreter.cc.o] Error 1
In file included from /mnt/c/workspace/nextpnr/gui/pythontab.cc:22:
/mnt/c/workspace/nextpnr/common/pybindings.h:24:10: fatal error: Python.h: No such file or directory
   24 | #include <Python.h>
      |          ^~~~~~~~~~
compilation terminated.
make[2]: *** [generated/gui/ice40/CMakeFiles/gui_ice40.dir/build.make:210: generated/gui/ice40/CMakeFiles/gui_ice40.dir/pythontab.cc.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:373: generated/gui/ice40/CMakeFiles/gui_ice40.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
```
This was despite my already having installed Python and dev libraries:

```
sudo apt-get install python3
sudo apt-get install python3-dev
sudo apt-get install libpython3.8-dev
```
Practically every google search result I found claimed one of the above commands was the solution to the missing `Python.h` problem. :/

and clearly I had several of those `Python.h` files scattered around my dirve:
[image]

I eventually found a [useful stackoverflow comment](https://stackoverflow.com/questions/11041299/python-h-no-such-file-or-directory) that mentioned using `python-config --cflags` and sure enough: it returned a bunch of Python2.7 references.

I tried to delete Python2.7 like this:

```
sudo apt-get remove python2.7
sudo apt-get autoremove
```
yet the problem persisted. I tried to start a fresh bash, add a known path to `Python.h` directory, and even reboot, all to no avail.

the solution? [Stackoverflow to the rescue again](https://stackoverflow.com/questions/9680420/looking-for-a-cmake-clean-command-to-clear-up-cmake-output)! From my `/mnt/c/workspace/nextpnr`:

```
rm cmake_install.cmake
rm CMakeCache.txt
rm -r ./CMakeFiles/
```
and voila! I was able to build `nextpnr`
```
cmake -DARCH=ice40  -DTRELLIS_INSTALL_PREFIX=/usr .
make -j$(nproc)
```

daveshah1, 2020-12-27:
I think this is just a case of CMake caching old stuff, not sure there is anything to be done on the nextpnr side. Thanks for the detailed instructions for anyone else who encounters this, though.

closed 2020-12-27 by daveshah1

sunnystormy, 2021-01-30:
I just encountered this problem, so I am *very* grateful for this detailed explanation. :)

ZhangPeterGree, 2021-11-18:
Yet! Nice.