YosysHQ/nextpnr #535: dyld: Library not loaded when running on Mac

jay20162016, 2020-12-08:
When I launch nextpnr, the following error shows up:
```
❯ /usr/local/bin/nextpnr-ice40
dyld: Library not loaded: @rpath/libQt5OpenGL.5.dylib
  Referenced from: /usr/local/bin/next❯ /usr/local/bin/nextpnr-ice40
dyld: Library not loaded: @rpath/libQt5OpenGL.5.dylib
  Referenced from: /usr/local/bin/nextpnr-ice40
  Reason: image not found
[1]    7618 abort      /usr/local/bin/nextpnr-ice40pnr-ice40
  Reason: image not found
[1]    7618 abort      /usr/local/bin/nextpnr-ice40
```

daveshah1, 2020-12-08:
I don't know enough about OSX to help with this issue, other than suggesting building without Qt (-DBUILD_GUI=no)

whitequark, 2020-12-08:
(Should the GUI perhaps be built as opt-in, not opt-out?)

daveshah1, 2020-12-08:
Seems like a reasonable idea, given the problems that Qt and OpenGL cause and the fact the GUI doesn't provide much value to a lot of users (personally, I very rarely use it) in its current form.

whitequark, 2020-12-08:
Do you want to update CMakeLists and README yourself, or would you like me to do it?

daveshah1, 2020-12-08:
I'm happy to do that