YosysHQ/nextpnr #546: Packaging for MSYS2

umarcor, 2020-12-27:
I'm finding it difficult to package nextpnr for MSYS2 cleanly. See: https://github.com/umarcor/nextpnr/blob/ci/msys2/msys2/PKGBUILD#L27-L69

- If `MSYS2_ARG_CONV_EXCL=-` is NOT used, `MINGW_PREFIX` is converted from `/mingw64` (or `/mingw32`) to an absolute path. Then, `mingw32-make DESTDIR="${pkgdir}" install` does not install the assets properly. Instead of placing them in `${pkgdir}/mingw64/bin`, it creates `${pkgdir}/path_at_build_time/msys64/mingw64/bin`. See https://github.com/umarcor/nextpnr/runs/1613032824?check_suite_focus=true#step:5:1871.

- Preprending cmake with `MSYS2_ARG_CONV_EXCL=-` is typically a solution. See, for instance, [prjtrellis](https://github.com/umarcor/MINGW-packages/blob/prjtrellis/mingw-w64-prjtrellis/PKGBUILD#L31) or [openFPGALoader](https://github.com/msys2/MINGW-packages/blob/master/mingw-w64-openFPGALoader/PKGBUILD#L32). Unfortunately, that seems not to work with nextpnr. See https://github.com/umarcor/nextpnr/runs/1614039582?check_suite_focus=true#step:5:1273: `FileNotFoundError: [Errno 2] No such file or directory: '/mingw64/share/icebox/timings_lp384.txt'`. Note that the file does exist, because it is found when the same path is resolved/converted.

I working around this issue by not using `MSYS2_ARG_CONV_EXCL=-` and copying the assets manually, instead of using `mingw32-make DESTDIR="${pkgdir}" install`:

```sh
  _bin="${pkgdir}${MINGW_PREFIX}/bin"
  mkdir -p "${_bin}"
  cp nextpnr-ice40.exe "${_bin}"
```

umarcor, 2020-12-28:
Using `MSYS2_ARG_CONV_EXCL` but setting `-DICESTORM_INSTALL_PREFIX` explicitly (https://github.com/umarcor/nextpnr/commit/6fbb0e09acd8ff164c3efc44b282bf1d1fa2de1c), allows the build to succeed and `mingw32-make DESTDIR="${pkgdir}" install` places the assets properly. Unfortunately, when `nextpnr-ice40 --help` is executed, it complains about `Qt5Widgets.dll` not being found. That's surprising, because the file exists, and it was found during the build. See: https://github.com/umarcor/nextpnr/runs/1616015155?check_suite_focus=true#step:5:8

```
Run which Qt5Widgets.dll
  which Qt5Widgets.dll
  nextpnr-ice40 --help
  
  shell: D:\a\_temp\msys\msys2.CMD {0}
  env:
    MSYSTEM: MINGW64
    
/mingw64/bin/Qt5Widgets.dll

D:/a/_temp/msys/msys64/mingw64/bin/nextpnr-ice40.exe: error while loading shared libraries: Qt5Widgets.dll: cannot open shared object file: No such file or directory

Error: Process completed with exit code 127.
```

umarcor, 2020-12-30:
FTR, the Qt5Widgets error was a red herring. The problem was boost missing in the runtime dependency list.

Some minutes ago, Yosys and nextpnr were accepted in MSYS2 :tada:. In the following days, both will be shown in [mingw-w64-x86_64-eda](https://packages.msys2.org/group/mingw-w64-x86_64-eda) and [mingw-w64-i686-eda](https://packages.msys2.org/group/mingw-w64-i686-eda).

However, the only enabled ARCH in the uploaded nextpnr package is `ice40`. I'd like to enable `ecp5` too. `prjtrellis` is packaged already and it works (see msys2/MINGW-packages#7509). Enabling `ecp5` on MINGW64 works too. Unfortunately, an error is produced on MINGW32:

```
2020-12-30T17:36:41.2036800Z Traceback (most recent call last):
2020-12-30T17:36:41.2533383Z   File "C:/_/mingw-w64-nextpnr/src/nextpnr/ecp5/trellis_import.py", line 645, in <module>
2020-12-30T17:36:41.2538567Z     main()
2020-12-30T17:36:41.2938824Z   File "C:/_/mingw-w64-nextpnr/src/nextpnr/ecp5/trellis_import.py", line 633, in main
2020-12-30T17:36:41.2950644Z     ddrg = pytrellis.make_dedup_chipdb(chip)
2020-12-30T17:36:41.3002726Z MemoryError: std::bad_alloc
2020-12-30T17:36:41.3099468Z make[2]: *** [ecp5/CMakeFiles/chipdb-ecp5-bbas.dir/build.make:103: ecp5/chipdb/chipdb-85k.bba] Error 1
2020-12-30T17:36:41.3100502Z make[2]: Leaving directory '/c/_/mingw-w64-nextpnr/src/nextpnr'
2020-12-30T17:36:41.3118116Z make[1]: *** [CMakeFiles/Makefile2:935: ecp5/CMakeFiles/chipdb-ecp5-bbas.dir/all] Error 2
```

daveshah1, 2020-12-30:
The database build needs more memory than a 32bit address space can provide. Use the bba file from the 64bit build.

umarcor, 2021-06-16:
I'm trying to update the nextpnr package on MSYS2. Unfortunately, the last commit I can build is 326b3488. This is the result with 9df05c4f: https://github.com/umarcor/MINGW-packages/runs/2837837400?check_suite_focus=true

```
2021-06-16T10:10:48.7339574Z /D/a/_temp/msys/msys64/mingw64/bin/g++.exe -march=x86-64 -mtune=generic -O2 -pipe -Wall -Wextra -Wno-unused-parameter -Wno-missing-field-initializers -Wno-array-bounds -fPIC -O3 -g -pipe -fopenmp -flto -fno-fat-lto-objects -pipe -Wl,--dynamicbase,--high-entropy-va,--nxcompat,--default-image-base-high -Wl,--whole-archive CMakeFiles/nextpnr-ice40-test.dir/objects.a -Wl,--no-whole-archive -o nextpnr-ice40-test.exe -Wl,--out-implib,libnextpnr-ice40-test.dll.a -Wl,--major-image-version,0,--minor-image-version,0  generated/3rdparty/googletest/libgtest_main.a /D/a/_temp/msys/msys64/mingw64/lib/libboost_filesystem-mt.dll.a /D/a/_temp/msys/msys64/mingw64/lib/libboost_program_options-mt.dll.a /D/a/_temp/msys/msys64/mingw64/lib/libboost_iostreams-mt.dll.a /D/a/_temp/msys/msys64/mingw64/lib/libboost_system-mt.dll.a /D/a/_temp/msys/msys64/mingw64/lib/libboost_thread-mt.dll.a /D/a/_temp/msys/msys64/mingw64/lib/libboost_regex-mt.dll.a /D/a/_temp/msys/msys64/mingw64/lib/libboost_chrono-mt.dll.a /D/a/_temp/msys/msys64/mingw64/lib/libboost_date_time-mt.dll.a /D/a/_temp/msys/msys64/mingw64/lib/libboost_atomic-mt.dll.a -lpthread generated/gui/ice40/libgui_ice40.a /D/a/_temp/msys/msys64/mingw64/lib/libQt5OpenGL.dll.a -lopengl32 -lglu32 generated/3rdparty/QtPropertyBrowser/src/libQtPropertyBrowser.a /D/a/_temp/msys/msys64/mingw64/lib/libpython3.8.dll.a generated/3rdparty/googletest/libgtest.a /D/a/_temp/msys/msys64/mingw64/lib/libQt5Widgets.dll.a /D/a/_temp/msys/msys64/mingw64/lib/libQt5Gui.dll.a /D/a/_temp/msys/msys64/mingw64/lib/libQt5Core.dll.a -lkernel32 -luser32 -lgdi32 -lwinspool -lshell32 -lole32 -loleaut32 -luuid -lcomdlg32 -ladvapi32 
2021-06-16T10:10:49.1632030Z D:/a/_temp/msys/msys64/mingw64/lib/gcc/x86_64-w64-mingw32/10.3.0/../../../../x86_64-w64-mingw32/bin/ld.exe: basectx.cc.obj (symbol from plugin):(.gnu.linkonce.t._ZN5boost10wrapexceptINS_17bad_function_callEED1Ev[_ZThn8_N5boost10wrapexceptINS_17bad_function_callEED1Ev]+0x0): multiple definition of `boost::wrapexcept<boost::bad_function_call>::~wrapexcept()'; hx1k.cc.obj (symbol from plugin):(.gnu.linkonce.t._ZN5boost10wrapexceptINS_17bad_function_callEED1Ev[_ZThn24_N5boost10wrapexceptINS_17bad_function_callEED1Ev]+0x0): first defined here
2021-06-16T10:10:49.2653026Z D:/a/_temp/msys/msys64/mingw64/lib/gcc/x86_64-w64-mingw32/10.3.0/../../../../x86_64-w64-mingw32/bin/ld.exe: basectx.cc.obj (symbol from plugin):(.gnu.linkonce.t._ZN5boost10wrapexceptINS_17bad_function_callEED1Ev[_ZThn8_N5boost10wrapexceptINS_17bad_function_callEED1Ev]+0x0): multiple definition of `non-virtual thunk to boost::wrapexcept<boost::bad_function_call>::~wrapexcept()'; hx1k.cc.obj (symbol from plugin):(.gnu.linkonce.t._ZN5boost10wrapexceptINS_17bad_function_callEED1Ev[_ZThn24_N5boost10wrapexceptINS_17bad_function_callEED1Ev]+0x0): first defined here
2021-06-16T10:10:49.3533443Z D:/a/_temp/msys/msys64/mingw64/lib/gcc/x86_64-w64-mingw32/10.3.0/../../../../x86_64-w64-mingw32/bin/ld.exe: basectx.cc.obj (symbol from plugin):(.gnu.linkonce.t._ZN5boost10wrapexceptINS_17bad_function_callEED1Ev[_ZThn8_N5boost10wrapexceptINS_17bad_function_callEED1Ev]+0x0): multiple definition of `non-virtual thunk to boost::wrapexcept<boost::bad_function_call>::~wrapexcept()'; hx1k.cc.obj (symbol from plugin):(.gnu.linkonce.t._ZN5boost10wrapexceptINS_17bad_function_callEED1Ev[_ZThn24_N5boost10wrapexceptINS_17bad_function_callEED1Ev]+0x0): first defined here
2021-06-16T10:10:53.8533726Z D:/a/_temp/msys/msys64/mingw64/lib/gcc/x86_64-w64-mingw32/10.3.0/../../../../x86_64-w64-mingw32/bin/ld.exe: place_common.cc.obj (symbol from plugin):(.gnu.linkonce.t._ZN5boost10wrapexceptINS_16bad_lexical_castEED1Ev[_ZThn8_N5boost10wrapexceptINS_16bad_lexical_castEED1Ev]+0x0): multiple definition of `boost::wrapexcept<boost::bad_lexical_cast>::~wrapexcept()'; command.cc.obj (symbol from plugin):(.gnu.linkonce.t._ZN5boost10wrapexceptINS_16bad_lexical_castEED1Ev[_ZThn32_N5boost10wrapexceptINS_16bad_lexical_castEED1Ev]+0x0): first defined here
2021-06-16T10:10:53.8547788Z D:/a/_temp/msys/msys64/mingw64/lib/gcc/x86_64-w64-mingw32/10.3.0/../../../../x86_64-w64-mingw32/bin/ld.exe: place_common.cc.obj (symbol from plugin):(.gnu.linkonce.t._ZN5boost10wrapexceptINS_16bad_lexical_castEED1Ev[_ZThn8_N5boost10wrapexceptINS_16bad_lexical_castEED1Ev]+0x0): multiple definition of `non-virtual thunk to boost::wrapexcept<boost::bad_lexical_cast>::~wrapexcept()'; command.cc.obj (symbol from plugin):(.gnu.linkonce.t._ZN5boost10wrapexceptINS_16bad_lexical_castEED1Ev[_ZThn32_N5boost10wrapexceptINS_16bad_lexical_castEED1Ev]+0x0): first defined here
2021-06-16T10:10:54.0002624Z D:/a/_temp/msys/msys64/mingw64/lib/gcc/x86_64-w64-mingw32/10.3.0/../../../../x86_64-w64-mingw32/bin/ld.exe: place_common.cc.obj (symbol from plugin):(.gnu.linkonce.t._ZN5boost10wrapexceptINS_16bad_lexical_castEED1Ev[_ZThn8_N5boost10wrapexceptINS_16bad_lexical_castEED1Ev]+0x0): multiple definition of `non-virtual thunk to boost::wrapexcept<boost::bad_lexical_cast>::~wrapexcept()'; command.cc.obj (symbol from plugin):(.gnu.linkonce.t._ZN5boost10wrapexceptINS_16bad_lexical_castEED1Ev[_ZThn32_N5boost10wrapexceptINS_16bad_lexical_castEED1Ev]+0x0): first defined here
[... cut ...]
2021-06-16T09:59:53.0145088Z make[2]: *** [CMakeFiles/nextpnr-ice40.dir/build.make:759: nextpnr-ice40.exe] Error 1
2021-06-16T09:59:53.0146037Z make[2]: Leaving directory '/c/_/mingw-w64-nextpnr/src/nextpnr'
2021-06-16T09:59:53.0156492Z make[1]: *** [CMakeFiles/Makefile2:299: CMakeFiles/nextpnr-ice40.dir/all] Error 2
2021-06-16T09:59:53.0157311Z make[1]: Leaving directory '/c/_/mingw-w64-nextpnr/src/nextpnr'
2021-06-16T09:59:53.0174781Z make: *** [Makefile:149: all] Error 2
```

Any help/hint is welcome.

gatecat, 2021-06-16:
Have you tried without LTO (`-DUSE_IPO=OFF`) - see #684

umarcor, 2021-06-16:
@gatecat, thanks a lot! ❤️  That worked: https://github.com/umarcor/MINGW-packages/actions/runs/942581055