YosysHQ/nextpnr #558: More efficient storage of wire/bel/pip names

daveshah1, 2021-01-29:
Based on recent discussions with @litghost (in the context of `getPipName`/`getPipByName` which are currently rarely used but might see more use in the future) and @TechnoMancer (in the context of nextpnr-generic) it is very easy for pip names in particular to cause a memory explosion due to needing to store the full name string for every pip.

The easiest solution would be to use a concatenated list of `IdString`s for object names (starting with wires/bels/pips but potentially extending to other objects too). Typically this would be a tile/site name in the first part(s) followed by an object name within the tile.

Outstanding issues:
 - due to different arches having different conventions, the delimiter will need to be flexible
 - efficient ways of storing these string lists will be needed (probably statically allocated for ≤4 strings i.e. 128 bits and heap allocated above that?) 
 - fast ways of parsing and generating them will be needed
 - functions like `nameOfPip` that return a `const char *` and are heavily used with `%s` in logging will need some rework because we no longer have a stored string we can just return a pointer too. Instead we'll need a string buffer pool like Yosys uses: https://github.com/YosysHQ/yosys/blob/708eb327a1e88f726b8d426dba35abc29339a634/kernel/log.cc#L605-L613
 - there will probably be a breaking change to the `nextpnr-generic` API to not require a full name for a bel/wire/pip but instead automatically create one given a tile prefix and object suffix

mithro, 2021-01-29:
@daveshah1 - I experimented with very efficient ways to store IdString type objects with a hierarchical nature at https://github.com/mithro/idstring -- It can store every FASM feature string for the complete Artix 50t in less than 40 kbytes of memory (see the examples section https://github.com/mithro/idstring/tree/master/examples/a50t_all_features).

closed 2021-02-03 by daveshah1

litghost, 2021-02-03:
This change decreases the memory requirements for doing archcheck on Xilinx Artix 50T from ~3-4 GB to ~2 GB.  I have a fix incoming that drops the memory requirement even lower to 166 MB.

PR #561 (2021-02-03): Use IdStringList for bel/wire/pip/group names
As bel/wire/pip/group names are generally comprised of a tile component (which itself contains many objects) and an object-in-tile component (which is the same across all tiles of the same type); storing the full name string for all objects, forever, is very inefficient.

This introduces a new type, `IdStringList`, comprising of a series of `IdString`s and an arch-specific delimiter, for these identifiers that avoids this waste.

Currently this PR contains a proof-of-concept RFC implementation for ECP5 only, excluding the GUI (which must be disable [... cut]