YosysHQ/nextpnr #569: Is polarity for unused REN on iCE40 1k correct?

cbiffle, 2021-02-04:
Hello!

Was reading the bitstream documentation for iCE40 as well as your sources, and I noticed [this](https://github.com/YosysHQ/nextpnr/blob/155e0b9c428aa32d1ca22d3679db6db50505b2a8/ice40/bitstream.cc#L774):

```c++
                if (ctx->args.type == ArchArgs::LP1K || ctx->args.type == ArchArgs::HX1K) {
                    set_config(ti, config.at(iey).at(iex), "IoCtrl.IE_" + std::to_string(iez), true);
                    set_config(ti, config.at(iey).at(iex), "IoCtrl.REN_" + std::to_string(iez), false);  // <---- this here
                } else {
                    set_config(ti, config.at(iey).at(iex), "IoCtrl.IE_" + std::to_string(iez), false);
                    set_config(ti, config.at(iey).at(iex), "IoCtrl.REN_" + std::to_string(iez), false);
                }
```

The Icestorm docs state that the REN polarity, like the IE polarity, is flipped on the 1k die vs the 8k. That would suggest to me that the `false` at the end of line 774 should be `true`. Have I missed something?

Thanks!

daveshah1, 2021-02-04:
It looks like arachne-pnr only inverts IE for the 1k, and I suspect the nextpnr behaviour was originally copied from that: https://github.com/YosysHQ/arachne-pnr/blob/master/src/place.cc#L1341-L1376

I will try and dig out icecube and some hardware tomorrow to double check.

cbiffle, 2021-02-04:
From the other cases where nextpnr sets `REN_*`, it seems like if my assertion were correct, pullups would be broken on the 1k. I feel like I would have noticed that? So now I am questioning myself too. I can test this pretty easily.

cbiffle, 2021-02-04:
Well! I manually constructed an empty bitstream for the ice40 hx1k, identified the I/O tile responsible for one of the LEDs on the icestick, and then identified the configuration bit responsible for its REN line. I can report that on this hx1k tq144, for `io_tile 13 12`, block 1, editing `B1[3]` to 0 causes the LED to light when the bitstream is loaded, 1 causes the LED to remain dark. The LEDs on the icestick board are driven from the anode, so lighting implies the pullup resistor is enabled.

My conclusion is that the REN bits on the hx1k are, in fact, _active low,_ and nextpnr is correct. I may have simply misread the Icestorm docs.

Sorry for the confusion!

closed 2021-02-04 by cbiffle