YosysHQ/nextpnr #593: nextpnr-gowin:  bram place error

curliph, 2021-02-20:
**I have two SPDB instances, the project fits well with gowin IDE.
After port to yosys+nextpnr-gowin,  got error  while pnr :**
----------------------------------------------------------------------------
Info: Device utilisation:
Info:                  SLICE:   657/ 1152    57%
Info:                    IOB:    15/  119    12%

Info: Placed 15 cells based on constraints.
ERROR: Unable to place cell 'u_ram.bram_sp_ram', no BELs remaining to implement cell type 'SDP'
------------------------------------------------------------------------------
**Resources usage reports:**
------------------------------------------------------------------------------
   Number of cells:                751
     DFF                            34
     DFFE                          180
     DFFR                            2
     DFFRE                          28
     DFFSE                          13
     GND                             1
     IBUF                            7
     LUT1                            2
     LUT2                           77
     LUT3                          149
     LUT4                          247
     OBUF                            8
     SDP                             2         // only 50% used
     VCC                             1
-------------------------------------------------------------------------------

gatecat, 2021-02-20:
My understanding is that BRAM aren't supported yet but @pepijndevos can confirm.

pepijndevos, 2021-02-21:
Yep, unfortunately BRAM is not yet supported. It's the first thing I plan to work on after fixing all the bugs, but at the current rate it could be a while... Contributions welcome of course. I'm always happy to help people get started with fuzzing. Adding BRAM should not be *that* hard actually. It's just a primitive with a few bits and ports. The initialization will require some refactoring but seems relatively straightforward.

curliph, 2021-02-22:
Got,  thanks for all your work!

closed 2021-02-22 by curliph