YosysHQ/nextpnr #610: Lattice ICE5LP packages: WLCS36 and CM36

esynr3z, 2021-03-07:
Trying to run some blinking on [iCE40 Ultra Breakout Board](http://www.latticesemi.com/products/developmentboardsandkits/ice40ultrabreakoutboard), where iCE5LP4K-SWG36 device in a 36-ball WLCSP package is installed.

The only available way I found is to use `--u4k --package sg48` keys combination. But there are some IO I want to use, that present in SWG36 and absent in SG48 package.

Are WLCS36 and CM36 packages unsupported for ICE40 Ultra family?

danielkucera, 2024-01-01:
I am having the same issue. I would adjust the code but I don't understand how the package is defined.
I created following definition in `/usr/share/fpga-icestorm/python/icebox.py`:
```
     "u4k-swg36": [
        ( "A2", 21, 21, 0),
        ( "A6",  6, 21, 0),
        ( "B1", 20,  0, 0),
        ( "B2", 22,  0, 1),
        ( "B4", 12,  0, 0),
        ( "B5", 12, 21, 1),
        ( "B6",  5, 21, 0),
        ( "C1", 22,  0, 0),
        ( "C2", 19,  0, 1),
        ( "C6",  4, 21, 0),
        ( "D1", 23,  0, 1),
        ( "D2", 20,  0, 1),
        ( "D5",  7,  0, 1),
        ( "D6",  7,  0, 0),
        ( "E1", 24,  0, 0),
        ( "E2", 21,  0, 1),
        ( "E3", 17,  0, 0),
        ( "E4", 13,  0, 0),
        ( "E5",  8,  0, 1),
        ( "E6",  6,  0, 1),
        ( "F1", 24,  0, 1),
        ( "F2", 23,  0, 0),
        ( "F3", 15,  0, 0),
        ( "F4", 12,  0, 1),
        ( "F5",  8,  0, 0),
        ( "F6",  6,  0, 0),
    ],
```
but it doesn't make any change in the behaviour of:
```
$ make
nextpnr-ice40 \
	--u4k --package swg36 \
	--pcf ../pcf/fomu-u4k.pcf \
	--json blink.json \
	--asc blink.asc
ERROR: Unsupported package 'swg36'.
0 warnings, 1 error
make: *** [Makefile:20: blink.asc] Error 255
```
Can anyone push me forward please?

mmicko, 2024-01-02:
Just tried at my machine to confirm nothing is missing, and you just need to `make install` icestorm first, and then build nextpnr-ice40 again and it will work just fine. You can make sure that it is fine if you grep chipdb-u4k.bba ( nextpnr/ice40/chipdb/chipdb-u4k.bba )  for swg36 after chipdb is regenerated

danielkucera, 2024-01-02:
Thank you, this was the missing piece - rebuilding nextpnr. I thought it is loaded somehow dynamically.
I've created PR for icestorm:
https://github.com/YosysHQ/icestorm/pull/320/files