YosysHQ/nextpnr #614: Error building on Ubuntu 20.04: cannot find -lpthreads

ColinAdreon, 2021-03-09:
Running `$ cmake -DARCH=ice40 -DPYTHONLIBS_VERSION_STRING=3.8 -DPYTHON_INCLUDE_DIR=/usr/include/python3.8 -DPYTHON_LIBRARY=/usr/lib/x86_64-linux-gnu/libpython3.8.so .` from the ```nextpnr``` root directory fails with the error
`
-- Found Boost: /usr/include (found version "1.71.0") found components: filesystem program_options iostreams system regex 
CMake Error at CMakeLists.txt:174 (add_subdirectory):
  The source directory

    /home/myusername/fpga/nextpnr/3rdparty/abseil-cpp

  does not contain a CMakeLists.txt file.

-- Found Boost: /usr/include (found version "1.71.0") found components: program_options filesystem system 
-- Configuring architecture: ice40
-- Enabled iCE40 devices: 384;1k;5k;u4k;8k
-- IceStorm install prefix: /usr/local
-- icebox data directory: /usr/local/share/icebox
-- Using iCE40 chipdb: /home/myusername/fpga/nextpnr/ice40/chipdb
-- Configuring incomplete, errors occurred!
`
Inspecting the log shows several errors relating to the pthread library, including the message in the title.  pthread is installed (version 2.31) so I don't think this is really the problem ([Stack Overflow](https://stackoverflow.com/questions/31948521/building-error-using-cmake-cannot-find-lpthreads) seems to agree).  The debug output shows an error with the abseil-cpp directory, which is empty.  I don't have Abseil on my system, but it seems unusual that if I needed a third-party dependency that it would need to be located within the nextpnr file structure.  Can anyone else make sense of this?

gatecat, 2021-03-09:
I think this is not a pthread issue but related to the abseil dependency that has been added. Do a `git submodule init` and `git submodule update` and try again.

ColinAdreon, 2021-03-10:
That appears to be the correct solution.  I would close this issue, but I think it would be beneficial for this information to be included in the readme first.

litghost, 2021-03-12:
> `git submodule init` and `git submodule update` and try again.

FYI, you can do:
```
git submodule update --init --recursive
```

On the clone you can do (with recent git clients):
```
git clone --recurse-submodules https://github.com/YosysHQ/nextpnr.git
```

closed 2021-03-15 by ColinAdreon

byllgrim, 2021-04-02:
> I would close this issue, but I think it would be beneficial for this information to be included in the readme first.

I think that would be good. I was about to give up trying to compile this tool, but I found this ticket.
Updating git submodules fixed it.