YosysHQ/nextpnr #622: Problems when executing the example of nextpnr- generic

TheRitzCarlton, 2021-03-14:
When I execute the command line:
yosys -p "tcl ../synth/synth_generic.tcl -json blinky.json" blinky.v
in the simple.sh in generic/example, I always get:
-- Running command `tcl ../synth/synth_generic.tcl -json blinky.json' --

Syntax error in command `tcl ../synth/synth_generic.tcl -json blinky.json':

    tcl <filename>

This command executes the tcl commands in the specified file.
Use 'yosys cmd' to run the yosys command 'cmd' from tcl.

The tcl command 'yosys -import' can be used to import all yosys
commands directly as tcl commands to the tcl shell. The yosys
command 'proc' is wrapped using the tcl command 'procs' in order
to avoid a name collision with the tcl builtin command 'proc'.

ERROR: Command syntax error: Extra argument.
> tcl ../synth/synth_generic.tcl -json blinky.json
>     ^
What's the problem with it and how to solve that? THX

gatecat, 2021-03-14:
Try updating Yosys, I know arguments to the tcl command were added a little while ago (you probably need git master Yosys rather than the last release.)

TheRitzCarlton, 2021-03-15:
THX.
I am trying building the yosys- master but currently meet some problem.
At the same time, I came up with an alternative : to manually synthesize the verilog file and use nextpnr-generic to place&route. I simply executed read_verilog, synth and write_json,(actually some commands in the tcl is missing in my current yosys downloaded from apt-get). But when executing p&r, the error occurred:

Info: Packing constants..
Info: Packing IOs..
Info: Packing LUT-FFs..
Info: Packing non-LUT FFs..
Info: Checksum: 0x6b12af7a

Info: Annotating ports with timing budgets for target frequency 12.00 MHz
Info: Checksum: 0x6b12af7a

Info: Device utilisation:
Info: 	       GENERIC_SLICE:     0/  800     0%
Info: 	         GENERIC_IOB:    10/   84    11%

Info: Placed 0 cells based on constraints.
ERROR: Unable to place cell '$abc$197$auto$blifparse.cc:286:parse_blif$198', no BELs remaining to implement cell type '$_NOR_'
0 warnings, 1 error

Does it also have something to do with the version of yosys or my command?

gatecat, 2021-03-15:
synth is not the correct command, it synthesises to gates and not the correct LUT primitives. Have a look at what the tcl script in the example is doing and follow those commands.

TheRitzCarlton, 2021-03-15:
Actually I tried it.
The first executable command in the tcl is：

yosys read_verilog -lib [file dirname [file normalize $argv0]]/prims.v

and I executed the following command in yosys：
read_verilog -lib ./prims.v

in the same directory as the tcl file.

The result is that the process was killed. What‘s wrong with it？

Furthermore, my current version of yosys doesn’t have “dfflegalize” command. Is it optional or compulsory?

gatecat, 2021-03-15:
I think your best option here is to find a way of getting recent Yosys and use that.

gatecat, 2021-03-17:
With YosysHQ/yosys#2661 resolved is this issue solved too?

TheRitzCarlton, 2021-03-18:
> With [YosysHQ/yosys#2661](https://github.com/YosysHQ/yosys/issues/2661) resolved is this issue solved too?

Yes.

closed 2021-03-18 by TheRitzCarlton