YosysHQ/nextpnr #671: Abseil built as shared library but not installed

zeldin, 2021-04-10:
Currently abseil is vendored and built by the cmake system.  However, it is built as a shared library and this shared library is not installed by make install.

This results in a situation where the installed binary can not be run:

```
nextpnr-ecp5: error while loading shared libraries: libabsl_hash.so: cannot open shared object file: No such file or directory
````

You need to either

* Link statically against abseil, so that everything is contained in the binary

or

* Don't vendor abseil, and require it to be installed at the system level instead

I don't think building a vendored abseil and installing it is a good idea, since it could conflict with a systemwide installation.

gatecat, 2021-04-10:
tbh abseil is giving far more problems than the small performance gain, for now I plan to remove it over the next few days and look at other options for fast map structures.

gatecat, 2021-04-15:
with abseil now disabled by default in #679, and a less problematic replacement will be found long term, I'm closing this issue

closed 2021-04-15 by gatecat