YosysHQ/nextpnr #675: nextpnr crashes on libc++abi.dylib: terminating with uncaught exception of type std::out_of_range: unordered_map::at: key not found

jay20162016, 2021-04-12:
Nextpnr crashes on my async circuit: https://github.com/jay20162016/fomu_async
Nextpnr works on other designs.

```
❯ nextpnr-ice40 --up5k --package uwg30 --asc sdf.asc --pcf ../hdl/fomu_pvt.pcf --json fomu.json --opt-timing --ignore-loops --debug --verbose

......

libc++abi.dylib: terminating with uncaught exception of type std::out_of_range: unordered_map::at: key not found
[1]    10760 abort      nextpnr-ice40 --up5k --package uwg30 --asc sdf.asc --pcf ../hdl/fomu_pvt.pcf
```
There is nothing atypical in the crash log, but the [full crash log (debug + verbose)](https://gist.github.com/jay20162016/035fafadb15dde6b6318b94bd1d230e7) is here

jay20162016, 2021-04-12:
```
Info: Device utilisation:
Info:            ICESTORM_LC:  1378/ 5280    26%
Info:           ICESTORM_RAM:     2/   30     6%
Info:                  SB_IO:     4/   96     4%
Info:                  SB_GB:     8/    8   100%
Info:           ICESTORM_PLL:     0/    1     0%
Info:            SB_WARMBOOT:     0/    1     0%
Info:           ICESTORM_DSP:     0/    8     0%
Info:         ICESTORM_HFOSC:     0/    1     0%
Info:         ICESTORM_LFOSC:     0/    1     0%
Info:                 SB_I2C:     0/    2     0%
Info:                 SB_SPI:     0/    2     0%
Info:                 IO_I3C:     0/    2     0%
Info:            SB_LEDDA_IP:     0/    1     0%
Info:            SB_RGBA_DRV:     1/    1   100%
Info:         ICESTORM_SPRAM:     0/    4     0%

terminate called after throwing an instance of 'std::out_of_range'
  what():  _Map_base::at

Program received signal SIGABRT, Aborted.
0x00007ffff74efef5 in raise () from /usr/lib/libc.so.6
(gdb) bt
#0  0x00007ffff74efef5 in raise () from /usr/lib/libc.so.6
#1  0x00007ffff74d9862 in abort () from /usr/lib/libc.so.6
#2  0x00007ffff787586a in __gnu_cxx::__verbose_terminate_handler () at /build/gcc/src/gcc/libstdc++-v3/libsupc++/vterminate.cc:95
#3  0x00007ffff7881d3a in __cxxabiv1::__terminate (handler=<optimized out>) at /build/gcc/src/gcc/libstdc++-v3/libsupc++/eh_terminate.cc:48
#4  0x00007ffff7881da7 in std::terminate () at /build/gcc/src/gcc/libstdc++-v3/libsupc++/eh_terminate.cc:58
#5  0x00007ffff788204e in __cxxabiv1::__cxa_throw (obj=obj@entry=0x5555642a4f80, tinfo=0x5555632b28c8 <typeinfo for std::out_of_range@GLIBCXX_3.4>, dest=0x7ffff7897f40 <std::out_of_range::~out_of_range()>)
    at /build/gcc/src/gcc/libstdc++-v3/libsupc++/eh_throw.cc:95
#6  0x00007ffff78785bd in std::__throw_out_of_range (__s=__s@entry=0x555563250ed6 "_Map_base::at") at /build/gcc/src/gcc/libstdc++-v3/src/c++11/functexcept.cc:82
#7  0x000055555561cf50 in std::__detail::_Map_base<int, std::pair<int const, nextpnr_ice40::TimingAnalyser::ArrivReqTime>, std::allocator<std::pair<int const, nextpnr_ice40::TimingAnalyser::ArrivReqTime> >, std::__detail::_Select1st, std::equal_to<int>, std::hash<int>, std::__detail::_Mod_range_hashing, std::__detail::_Default_ranged_hash, std::__detail::_Prime_rehash_policy, std::__detail::_Hashtable_traits<false, false, true>, true>::at (this=<optimized out>, __k=<optimized out>)
    at /usr/include/c++/10.2.0/bits/hashtable_policy.h:766
#8  0x0000555555614074 in std::unordered_map<int, nextpnr_ice40::TimingAnalyser::ArrivReqTime, std::hash<int>, std::equal_to<int>, std::allocator<std::pair<int const, nextpnr_ice40::TimingAnalyser::ArrivReqTime> > >::at (__k=@0x7fffffffca24: 0,
    this=<optimized out>) at /usr/include/c++/10.2.0/bits/unordered_map.h:1001
#9  nextpnr_ice40::TimingAnalyser::set_arrival_time (this=this@entry=0x7fffffffd0f8, target=..., domain=<optimized out>, arrival=..., path_length=1, prev=...) at ../common/timing.cc:289
#10 0x0000555555616b40 in nextpnr_ice40::TimingAnalyser::walk_forward (this=0x7fffffffd0f8) at ../common/timing.cc:358
#11 0x0000555555617da4 in nextpnr_ice40::TimingAnalyser::run (this=0x7fffffffd0f8) at ../common/timing.cc:46
#12 nextpnr_ice40::TimingAnalyser::setup (this=<optimized out>) at ../common/timing.cc:39
#13 0x000055555563b1f5 in nextpnr_ice40::HeAPPlacer::HeAPPlacer (cfg=..., ctx=<optimized out>, this=0x7fffffffcf80) at ../common/placer_heap.cc:147
#14 nextpnr_ice40::placer_heap (cfg=..., ctx=<optimized out>) at ../common/placer_heap.cc:1822
#15 nextpnr_ice40::Arch::place (this=0x5555632d4710) at ../ice40/arch.cc:689
#16 0x00005555556b0f53 in nextpnr_ice40::CommandHandler::executeMain(std::unique_ptr<nextpnr_ice40::Context, std::default_delete<nextpnr_ice40::Context> >) [clone .constprop.0] (this=this@entry=0x7fffffffde80,
    ctx=std::unique_ptr<nextpnr_ice40::Context> = {...}) at ../common/command.cc:377
#17 0x0000555555595253 in nextpnr_ice40::CommandHandler::exec (this=0x7fffffffde80) at ../common/command.cc:449
#18 main (argc=<optimized out>, argv=<optimized out>) at ../ice40/main.cc:281
```
I think this is the gdb output (Thanks Echo on discord), not sure tho

jay20162016, 2021-04-12:
#609  seems to be the pr that breaks this

gatecat, 2021-04-13:
Thanks for the report and test case, this turned out to be quite an interesting issue. See #676 for a fix

PR #676 (2021-04-13): timing: Fix domain init when loops are present
Fixes #675
closed 2021-04-13 by gatecat