YosysHQ/nextpnr #684: CMake IPO breaks compilation on MinGW

sylefeb, 2021-04-20:
When compiling for MSYS2 / MinGW64 (under Windows 10), USE_IPO is set to ON in the CMake build. This results in linker errors seemingly related to boost, of the likes of:
```
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/10.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: basectx.cc.obj (symbol from plugin):(.gnu.linkonce.t._ZN5boost10wrapexceptINS_17bad_function_callEED1Ev[_ZThn8_N5boost10wrapexceptINS_17bad_function_callEED1Ev]+0x0): multiple definition of `boost::wrapexcept<boost::bad_function_call>::~wrapexcept()'; archcheck.cc.obj (symbol from plugin):(.gnu.linkonce.t._ZN5boost10wrapexceptINS_17bad_function_callEED1Ev[_ZThn24_N5boost10wrapexceptINS_17bad_function_callEED1Ev]+0x0): first defined here
```
(obtained with `g++.exe (Rev10, Built by MSYS2 project) 10.2.0`, Boost `1.75`, commit 3fd1ee7757356660c7f440705553d345837eaed5

Force disabling USE_IPO from the cmake command line (`-DUSE_IPO=OFF`) fixes the issue. The IPO check might have to be modified to exclude MinGW.

gatecat, 2021-04-20:
@mmicko are you still doing MinGW builds? have you seen this?

mmicko, 2021-04-20:
@gatecat Yes I have issue with IPO, even when doing cross compiling and got large number of cases like above, so I have used USE_IPO=OFF

@sylefeb Believe disabling IPO is best bet for mingw builds, this is something that is up to mingw GCC compiler version so not much we can help there

cr1901, 2021-05-16:
@sylefeb IPO worked for me as of the linked PR (to disable it :)): https://github.com/YosysHQ/nextpnr/pull/665

I also have the same compiler version and boost version as you. I can't recompile right now, but I'll do a clean build sometime in the next few days at my current commit (883ece60 plus a few more in my personal branch).

If I can still build w/ IPO, I'll do a bisect and see what happens.

cr1901, 2021-06-01:
@sylefeb I was mistaken; I can't get IPO to work on MinGW at all. For VPR, I recall I _was_ able to get IPO to work, so I guess I conflated VPR with `nextpnr` in this case :D.

From my testing, IPO has been broken on MinGW since introduction in d5021e7. Perhaps it should be disabled unconditionally w/ a warning on MinGW, to be reenabled when someone feels brave enough to try to fix it? 

Doesn't seem like there's [high priority](https://stackoverflow.com/questions/60591742/mingw-gcc-multiple-definition-of-destructor-and-non-virtual-thunk#comment107212321_60591742) to fix it on the binutils side. I don't even known what a `non-virtual think` is ._.:

```
[56/56] Linking CXX executable nextpnr-ice40.exe
FAILED: nextpnr-ice40.exe
cmd.exe /C "cd . && C:\msys64\mingw64\bin\c++.exe -Wall -Wextra -Wno-unused-parameter -Wno-missing-field-initializers -Wno-array-bounds -fPIC -O3 -g -pipe -flto -fno-fat-lto-objects  @CMakeFiles\nextpnr-ice40.rsp -o nextpnr-ice40.exe -Wl,--major-image-version,0,--minor-image-version,0  && cd ."
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/10.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: CMakeFiles/nextpnr-ice40.dir/common/basectx.cc.obj (symbol from plugin):(.gnu.linkonce.t._ZN5boost10wrapexceptINS_17bad_function_callEED1Ev[_ZThn8_N5boost10wrapexceptINS_17bad_function_callEED1Ev]+0x0): multiple definition of `boost::wrapexcept<boost::bad_function_call>::~wrapexcept()'; CMakeFiles/nextpnr-ice40.dir/common/archcheck.cc.obj (symbol from plugin):(.gnu.linkonce.t._ZN5boost10wrapexceptINS_17bad_function_callEED1Ev[_ZThn24_N5boost10wrapexceptINS_17bad_function_callEED1Ev]+0x0): first defined here
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/10.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: CMakeFiles/nextpnr-ice40.dir/common/basectx.cc.obj (symbol from plugin):(.gnu.linkonce.t._ZN5boost10wrapexceptINS_17bad_function_callEED1Ev[_ZThn8_N5boost10wrapexceptINS_17bad_function_callEED1Ev]+0x0): multiple definition of `non-virtual thunk to boost::wrapexcept<boost::bad_function_call>::~wrapexcept()'; CMakeFiles/nextpnr-ice40.dir/common/archcheck.cc.obj (symbol from plugin):(.gnu.linkonce.t._ZN5boost10wrapexceptINS_17bad_function_callEED1Ev[_ZThn24_N5boost10wrapexceptINS_17bad_function_callEED1Ev]+0x0): first defined here
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/10.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: CMakeFiles/nextpnr-ice40.dir/common/basectx.cc.obj (symbol from plugin):(.gnu.linkonce.t._ZN5boost10wrapexceptINS_17bad_function_callEED1Ev[_ZThn8_N5boost10wrapexceptINS_17bad_function_callEED1Ev]+0x0): multiple definition of `non-virtual thunk to boost::wrapexcept<boost::bad_function_call>::~wrapexcept()'; CMakeFiles/nextpnr-ice40.dir/common/archcheck.cc.obj (symbol from plugin):(.gnu.linkonce.t._ZN5boost10wrapexceptINS_17bad_function_callEED1Ev[_ZThn24_N5boost10wrapexceptINS_17bad_function_callEED1Ev]+0x0): first defined here
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/10.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: CMakeFiles/nextpnr-ice40.dir/common/place_common.cc.obj (symbol from plugin):(.gnu.linkonce.t._ZN5boost10wrapexceptINS_16bad_lexical_castEED1Ev[_ZThn8_N5boost10wrapexceptINS_16bad_lexical_castEED1Ev]+0x0): multiple definition of `boost::wrapexcept<boost::bad_lexical_cast>::~wrapexcept()'; CMakeFiles/nextpnr-ice40.dir/common/command.cc.obj (symbol from plugin):(.gnu.linkonce.t._ZN5boost10wrapexceptINS_16bad_lexical_castEED1Ev[_ZThn32_N5boost10wrapexceptINS_16bad_lexical_castEED1Ev]+0x0): first defined here
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/10.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: CMakeFiles/nextpnr-ice40.dir/common/place_common.cc.obj (symbol from plugin):(.gnu.linkonce.t._ZN5boost10wrapexceptINS_16bad_lexical_castEED1Ev[_ZThn8_N5boost10wrapexceptINS_16bad_lexical_castEED1Ev]+0x0): multiple definition of `non-virtual thunk to boost::wrapexcept<boost::bad_lexical_cast>::~wrapexcept()'; CMakeFiles/nextpnr-ice40.dir/common/command.cc.obj (symbol from plugin):(.gnu.linkonce.t._ZN5boost10wrapexceptINS_16bad_lexical_castEED1Ev[_ZThn32_N5boost10wrapexceptINS_16bad_lexical_castEED1Ev]+0x0): first defined here
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/10.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: CMakeFiles/nextpnr-ice40.dir/common/place_common.cc.obj (symbol from plugin):(.gnu.linkonce.t._ZN5boost10wrapexceptINS_16bad_lexical_castEED1Ev[_ZThn8_N5boost10wrapexceptINS_16bad_lexical_castEED1Ev]+0x0): multiple definition of `non-virtual thunk to boost::wrapexcept<boost::bad_lexical_cast>::~wrapexcept()'; CMakeFiles/nextpnr-ice40.dir/common/command.cc.obj (symbol from plugin):(.gnu.linkonce.t._ZN5boost10wrapexceptINS_16bad_lexical_castEED1Ev[_ZThn32_N5boost10wrapexceptINS_16bad_lexical_castEED1Ev]+0x0): first defined here
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/10.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: CMakeFiles/nextpnr-ice40.dir/common/placer_heap.cc.obj (symbol from plugin):(.gnu.linkonce.t._ZN5boost10wrapexceptISt9bad_allocED1Ev[_ZThn8_N5boost10wrapexceptISt9bad_allocED1Ev]+0x0): multiple definition of `boost::wrapexcept<std::bad_alloc>::~wrapexcept()'; CMakeFiles/nextpnr-ice40.dir/common/archcheck.cc.obj (symbol from plugin):(.gnu.linkonce.t._ZN5boost10wrapexceptISt9bad_allocED1Ev[_ZThn16_N5boost10wrapexceptISt9bad_allocED1Ev]+0x0): first defined here
[... cut ...]
C:/msys64/mingw64/bin/../lib/gcc/x86_64-w64-mingw32/10.2.0/../../../../x86_64-w64-mingw32/bin/ld.exe: CMakeFiles/nextpnr-ice40.dir/common/placer_heap.cc.obj (symbol from plugin):(.gnu.linkonce.t._ZN5boost10wrapexceptISt9bad_allocED1Ev[_ZThn8_N5boost10wrapexceptISt9bad_allocED1Ev]+0x0): multiple definition of `non-virtual thunk to boost::wrapexcept<std::bad_alloc>::~wrapexcept()'; CMakeFiles/nextpnr-ice40.dir/common/archcheck.cc.obj (symbol from plugin):(.gnu.linkonce.t._ZN5boost10wrapexceptISt9bad_allocED1Ev[_ZThn16_N5boost10wrapexceptISt9bad_allocED1Ev]+0x0): first defined here
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
```