YosysHQ/nextpnr #691: ERROR: Found two arcs with same sink wire X24/Y31/io_global/latch

jmlzone, 2021-04-30:
Hi,
  I cant seem to find any error in the Verilog into or out of yosys showing the tow nets it it claims are going to the same sink.  They are all competently unique nets going to similar but unique instances of SB_IO. 

Some suggestions on how to debug.

Some background.
yosys and nextpnr built on raspberry pi 4.  (they run really fast!).
I am unable to get the GUI to build due to problems with qt5/gl3.2 called out in the ImGuiRenderer.h and ImGuiRenderer.cpp.

I am good with verilog design and have some competence in C/python, but I have no idea how to debug this.

thanks,
James

jmlzone, 2021-04-30:
Some more research/trial and error.
I have 4 unique SB_IO where I was using 4 unique signals to each of the latch_input inputs.
fiddling with the pin assignments or verilog. it would complain about a different pair of the signals. 
I can remove all but one of the signals to LATCH_INPUT_VALUE and I can run.

The IO diagrams in the lattice documents appear to show that the LATCH_INPUT_VALUE signals are unique (vs the input clock and output clock shown as shared), but a not says the freeze signal is common to the bank. 

Since the 5Ksg48 has three banks of IO i tried using 3 of them one in each bank, it still complains:
 ERROR: Found two arcs with same sink wire X24/Y0/io_global/latch: hold_south (0) vs hold_north (0)

but 'north' is on pin 46, on bank 2, and south is on pin, 19 which is bank 1.

jmlzone, 2021-04-30:
PS for the purposes of my FPGA development I have coded around this error by adding my own secondary registers.
if this is just user error/poor latice documentation then just close.  If we need to reproduce/.debug I have a copy of my files.

gatecat, 2021-05-02:
Do you have access to Radiant or iCEcube? It would be useful to know whether or not they can build the design. A complete set of design files to test with would be useful, too.

jmlzone, 2021-05-03:
Thanks for the comment @gatecat I currently have only yosys/nextpnr.  I do have a tar file with the state of my design at the time of the error. I will take the time to un-tar and then clean up to just what's needed to reproduce and I can make that available.

  if there are other tools I can download/build on Rpi I can do that as well to help with the debug.

Since there is some interest n the debug, I will work on preparing the files to reproduce in the next few days.

jmlzone, 2021-05-03:
[nexpnr_691.tar.gz](https://github.com/YosysHQ/nextpnr/files/6415908/nexpnr_691.tar.gz)

tar gzip file with design files and scripts.

README for files associated with issue 691
verilog/ contains the source design files.
scripts/ contains the scripts (and a set of yosys results)
        etargetu5k.pcf_bank0
	     -- initial pcf file with all usage of the latch in bank0
	etargetu5k.pcf
	     -- north on pin 46 (bank2)
	     -- south on pin 19 (bank1)
	     -- east on 26 (bank0)
	     -- west on 31 (bank 0 but latch not specified) 
Scripts here for reproducing without touching the the json synth results in scripts:

syn.sh -- run yosys for synthesys
pnr.sh -- run nexpnr