YosysHQ/nextpnr #699: Nextpnr gives wrong output - changing "assign a = b" to direct wire (in instantiation) works

jay20162016, 2021-05-09:
Hello! My design at https://github.com/jay20162016/fomu_async is not working again. 
Uncommenting the line with the "works" comment (and commenting the "doesn't work" lines) makes it work.
Similarly, the "doesn't work" lines don't work.
Here is the relevant part of the code:
```verilog
...
  swap #(.Rpol(Rpol), .N(N), .NATIVE(NATIVE)) main (
    r_i, a_i, d_i,
    r1_i, a1_i, d1_i,
    ctl_a, ctl_b, actl_i,
    r_o, a_o, d_o,
    // rt_o, rt_o, dt_o, // works
    rt_o, at_o, dt_o,
    rst
    );

  sink main_sink (
    rt_o, at_o
    ); // doesn't work
  // assign at_o = rt_o; // doesn't work
  // assign rt_o = at_o;
...
```

I have already verified that in yosys, the output is the same, except for a few extra wires (the typical stuff).

Thanks in advance!

jay20162016, 2021-05-09:
To test, run `fomu.sh` in the `synthesis` directory (do note to replace the programming logic with your own; I use a fomu with its dfu programming logic). 
[image]
Check the boxes as so - do note that you must check the box in row 2 before checking the boxes in row 1.

[image]
Then uncheck all boxes like in the picture (besides reset in the right corner). Note that column 3 should not change (this image is of the buggy variant).

jay20162016, 2021-05-09:
No, altering --seed doesn't change anything.

jay20162016, 2021-05-10:
> I have already verified that in yosys, the output is the same, except for a few extra wires (the typical stuff).

Ok, specifically only that module; adding the module to everything else results in a big difference.
So it might be a yosys bug.

jay20162016, 2021-05-10:
https://gist.github.com/jay20162016/1aeaa09e75ac65907619ca580fa3ed8d

jay20162016, 2021-05-11:
Should be a yosys bug - see https://github.com/YosysHQ/yosys/issues/2751

gatecat, 2021-05-12:
I'm not sure if there is a bug in either tool. It would definitely be best if you could provide a more concrete example of what is going wrong.

Note that small changes in a design can definitely cause big changes in synthesis tool output, this is simply a consequence of how Yosys works and not  a bug in its own right. Given the nature of asynchronous logic, it's possible that is what is breaking your design.

closed 2021-05-12 by gatecat

jay20162016, 2021-05-12:
But the small, functionally identical changes change the synthesis tool's output completely in a way that functionally affects how the design works.