YosysHQ/nextpnr #716: Fluctuating QoR and inconsistent runs despite constant seed

smunaut, 2021-06-01:
I've been experiencing some issues with latest nextpnr for the ice40. (Didn't test other arches).
The QoR seemed way worse than it was for my old build (back from Dec 2020) and I also noticed that despite specifying a --seed, the results could be different run to run. Not always different, sometime it would take 3-4 runs to notice a change, but with the old binary I could run it 30 times and always get identical results.

Attached is the design I used and a test script :

[qor-test.tar.gz](https://github.com/YosysHQ/nextpnr/files/6574976/qor-test.tar.gz)

I ran the test both with the latest nightly from oss-cad-suite ( oss-cad-suite-linux-x64-20210601.tgz ) and from fpga-toolchain ( fpga-toolchain-linux_x86_64-nightly-20210601.tar.xz ).  Now both of theses do produce results consistent with themselves (i.e. for the same seed you always get the same result ). However the QoR is drastically different.

Out of 64 runs/seeds, the oss-cad-suite results in 5 timing miss and they are barely missing.
The fpga-toolchain build results in 35 miss and some by a very wide margin. ( Raw results below)

Although the non-consistent behavior is not shown with the pre-built toolchain, I somehow suspect the 2 issues being linked. Maybe something with ASLR not being present in the binary build or something like that ...

```
$ ./test.sh ../oss-cad-suite/bin/nextpnr-ice40 
Run for ../oss-cad-suite/bin/nextpnr-ice40 (447288b6bec573559313bb7efe51994568802435)
Warning: Max frequency for clock 'clk': 47.03 MHz (FAIL at 48.00 MHz)
1 warning, 0 errors
Warning: Max frequency for clock 'clk': 47.03 MHz (FAIL at 48.00 MHz)
1 warning, 0 errors
Warning: Max frequency for clock 'clk': 47.98 MHz (FAIL at 48.00 MHz)
1 warning, 0 errors
Warning: Max frequency for clock 'clk': 47.03 MHz (FAIL at 48.00 MHz)
1 warning, 0 errors
Warning: Max frequency for clock 'clk': 47.92 MHz (FAIL at 48.00 MHz)
1 warning, 0 errors
Errors: 5

$ ./test.sh ../fpga-toolchain/bin/nextpnr-ice40 
Run for ../fpga-toolchain/bin/nextpnr-ice40 (5076f867048bd8c1aac491ec88e2faa633586f89)
Warning: Max frequency for clock 'clk': 46.79 MHz (FAIL at 48.00 MHz)
1 warning, 0 errors
Warning: Max frequency for clock 'clk': 47.18 MHz (FAIL at 48.00 MHz)
1 warning, 0 errors
Warning: Max frequency for clock 'clk': 47.62 MHz (FAIL at 48.00 MHz)
Warning: Max frequency for clock 'clk': 47.30 MHz (FAIL at 48.00 MHz)
1 warning, 0 errors
1 warning, 0 errors
Warning: Max frequency for clock 'clk': 47.53 MHz (FAIL at 48.00 MHz)
Warning: Max frequency for clock 'clk': 45.42 MHz (FAIL at 48.00 MHz)
1 warning, 0 errors
1 warning, 0 errors
Warning: Max frequency for clock 'clk': 46.48 MHz (FAIL at 48.00 MHz)
1 warning, 0 errors
Warning: Max frequency for clock 'clk': 46.36 MHz (FAIL at 48.00 MHz)
Warning: Max frequency for clock 'clk': 45.72 MHz (FAIL at 48.00 MHz)
1 warning, 0 errors
1 warning, 0 errors
Warning: Max frequency for clock 'clk': 46.48 MHz (FAIL at 48.00 MHz)
1 warning, 0 errors
Warning: Max frequency for clock 'clk': 43.36 MHz (FAIL at 48.00 MHz)
Warning: Max frequency for clock 'clk': 46.77 MHz (FAIL at 48.00 MHz)
Warning: Max frequency for clock 'clk': 43.61 MHz (FAIL at 48.00 MHz)
1 warning, 0 errors
Warning: Max frequency for clock 'clk': 46.94 MHz (FAIL at 48.00 MHz)
Warning: Max frequency for clock 'clk': 47.59 MHz (FAIL at 48.00 MHz)
1 warning, 0 errors
1 warning, 0 errors
1 warning, 0 errors
Warning: Max frequency for clock 'clk': 45.56 MHz (FAIL at 48.00 MHz)
1 warning, 0 errors
Warning: Max frequency for clock 'clk': 47.18 MHz (FAIL at 48.00 MHz)
Warning: Max frequency for clock 'clk': 47.30 MHz (FAIL at 48.00 MHz)
1 warning, 0 errors
1 warning, 0 errors
1 warning, 0 errors
Warning: Max frequency for clock 'clk': 45.07 MHz (FAIL at 48.00 MHz)
Warning: Max frequency for clock 'clk': 45.20 MHz (FAIL at 48.00 MHz)
Warning: Max frequency for clock 'clk': 44.77 MHz (FAIL at 48.00 MHz)
1 warning, 0 errors
1 warning, 0 errors
Warning: Max frequency for clock 'clk': 44.36 MHz (FAIL at 48.00 MHz)
Warning: Max frequency for clock 'clk': 47.68 MHz (FAIL at 48.00 MHz)
1 warning, 0 errors
1 warning, 0 errors
Warning: Max frequency for clock 'clk': 47.18 MHz (FAIL at 48.00 MHz)
Warning: Max frequency for clock 'clk': 43.01 MHz (FAIL at 48.00 MHz)
1 warning, 0 errors
Warning: Max frequency for clock 'clk': 45.80 MHz (FAIL at 48.00 MHz)
1 warning, 0 errors
1 warning, 0 errors
1 warning, 0 errors
Warning: Max frequency for clock 'clk': 47.24 MHz (FAIL at 48.00 MHz)
Warning: Max frequency for clock 'clk': 47.03 MHz (FAIL at 48.00 MHz)
1 warning, 0 errors
1 warning, 0 errors
Warning: Max frequency for clock 'clk': 45.89 MHz (FAIL at 48.00 MHz)
Warning: Max frequency for clock 'clk': 45.31 MHz (FAIL at 48.00 MHz)
1 warning, 0 errors
Warning: Max frequency for clock 'clk': 47.18 MHz (FAIL at 48.00 MHz)
1 warning, 0 errors
Warning: Max frequency for clock 'clk': 45.37 MHz (FAIL at 48.00 MHz)
1 warning, 0 errors
Warning: Max frequency for clock 'clk': 46.06 MHz (FAIL at 48.00 MHz)
1 warning, 0 errors
1 warning, 0 errors
Warning: Max frequency for clock 'clk': 47.62 MHz (FAIL at 48.00 MHz)
1 warning, 0 errors
Warning: Max frequency for clock 'clk': 46.94 MHz (FAIL at 48.00 MHz)
1 warning, 0 errors
Errors: 35
```

smunaut, 2021-06-01:
Oh, I forgot to mention that I tried bissecting. Now the results aren't great because a lot of commits don't build at all or result in crashes, probably some intermediate commits from dev branches, but it seemed to point to ebc2527368d920ea3c40a9ca83f73df242785044 and the few commits around it as the cause.

At the very least fba71bd182151713455c8d1cf0abefea9cf59831 seems to be the latest commit in the "linear" history that I can build, doesn't crash and doesn't exhibit any bad QoR or any consistency issue.

smunaut, 2021-06-01:
Tried running valgrin on latest master ( 24ae205f20f0e1a0326e48002ab14d5bacfca1ef ). It shows decision based on unitialized values ( which it doesn't for older / known working nextpnr ).

See https://pastebin.com/2ebMzNUG for a report.

Using VGDB to break on the first such instance :

```
==29427== Conditional jump or move depends on uninitialised value(s)
==29427==    at 0x1B1615: nextpnr_ice40::TimingAnalyser::run() (stl_algobase.h:203)
==29427==    by 0x1B24C6: nextpnr_ice40::TimingAnalyser::setup() (timing.cc:39)
==29427==    by 0x1DAF7C: nextpnr_ice40::placer_heap(nextpnr_ice40::Context*, nextpnr_ice40::PlacerHeapCfg) (placer_heap.cc:147)
==29427==    by 0x18B640: nextpnr_ice40::Arch::place() (arch.cc:689)
==29427==    by 0x13F1B3: nextpnr_ice40::CommandHandler::executeMain(std::unique_ptr<nextpnr_ice40::Context, std::default_delete<nextpnr_ice40::Context> >) [clone .constprop.0] (command.cc:388)
==29427==    by 0x1342E5: UnknownInlinedFun (command.cc:465)
==29427==    by 0x1342E5: main (main.cc:281)
==29427== 
==29427== (action on error) vgdb me ... 
```

```
(gdb) bt
#0  0x00000000001b1615 in std::min<float> (__b=<optimized out>, __a=<synthetic pointer>: 1.00000334)
    at /usr/lib/gcc/x86_64-pc-linux-gnu/9.3.0/include/g++-v9/bits/stl_algobase.h:198
#1  nextpnr_ice40::TimingAnalyser::compute_criticality (this=<optimized out>)
    at /home/tnt/projects/fpga/toolchain/nextpnr/common/timing.cc:480
#2  nextpnr_ice40::TimingAnalyser::run (this=<optimized out>) at /home/tnt/projects/fpga/toolchain/nextpnr/common/timing.cc:49
#3  0x00000000001b24c7 in nextpnr_ice40::TimingAnalyser::setup (this=0x1ffeffe7d8)
    at /home/tnt/projects/fpga/toolchain/nextpnr/common/timing.cc:39
#4  0x00000000001daf7d in nextpnr_ice40::HeAPPlacer::HeAPPlacer (cfg=..., ctx=0x10e70140, this=0x1ffeffe660)
    at /home/tnt/projects/fpga/toolchain/nextpnr/common/placer_heap.cc:147
#5  nextpnr_ice40::placer_heap (ctx=0x10e70140, cfg=...) at /home/tnt/projects/fpga/toolchain/nextpnr/common/placer_heap.cc:1781
#6  0x000000000018b641 in nextpnr_ice40::Arch::place (this=0x10e70140)
    at /home/tnt/projects/fpga/toolchain/nextpnr/common/basectx.h:161
#7  0x000000000013f1b4 in nextpnr_ice40::CommandHandler::executeMain(std::unique_ptr<nextpnr_ice40::Context, std::default_delete<nextpnr_ice40::Context> >) [clone .constprop.0] (this=0x1ffefff740, ctx=...)
    at /home/tnt/projects/fpga/toolchain/nextpnr/common/command.cc:388
#8  0x00000000001342e6 in nextpnr_ice40::CommandHandler::exec (this=0x1ffefff740)
    at /usr/lib/gcc/x86_64-pc-linux-gnu/9.3.0/include/g++-v9/bits/hashtable_policy.h:1296
#9  main (argc=<optimized out>, argv=<optimized out>) at /home/tnt/projects/fpga/toolchain/nextpnr/ice40/main.cc:281
```

gatecat, 2021-06-01:
#717 fixes the valgrind issue. I haven't looked at the QoR issue itself yet though.

smunaut, 2021-06-01:
As far as I'm concerned this is fixed by #717

closed 2021-06-01 by smunaut