YosysHQ/nextpnr #732: Output placer and router details files

y8j31, 2021-06-18:
Hi,
I am using the branch 'xc7' to place and route on my own FPGA fabric, I know this version is out of date. But is there any luck this version of nextpnr can output the placer and router details for debug purpose? For example, a file inlucdes the locations description after placing, and a file includes which net is left to be routed in the router.

Kind regards

gatecat, 2021-06-18:
Hi!

This is an example Python script that dumps placement information that should be usable on that branch: https://github.com/YosysHQ/nextpnr/blob/xc7/python/dump_design.py

However, I'd definitely recommend you think about updating at some point - that branch is about 2.5 years old in a codebase that's 3 years old so there have been many improvements since then - a much faster analytical placer, countless bug fixes, a new router, more deterministic behaviour, a JSON backend that includes place and route information, etc.

y8j31, 2021-06-21:
> Hi!
>
> This is an example Python script that dumps placement information that should be usable on that branch: https://github.com/YosysHQ/nextpnr/blob/xc7/python/dump_design.py
>
> However, I'd [...]

Hi gatecat,
Thanks a lot for your reply. Yes, I am thinking about the update once I have the time. Do you have any clues which branch I can start with to update and apply on my own fabric?
Cheers

gatecat, 2021-06-21:
I would strongly recommend starting from the `master` branch. Although they are both somewhat more complicated than you are likely to need, both `nexus` and `mistral` are good examples of new arches demonstrating recent design patterns (like not packing LUTs and FFs together but having validity checks during placement for this). `generic` is a simpler base that might be more relevant from a database point of view.

y8j31, 2021-09-30:
> I would strongly recommend starting from the `master` branch. Although they are both somewhat more complicated than you are likely to need, both `nexus` and `mistral` are good examples of new arches [...]

Can the GENERIC_SLICE in ```generic``` be modified into a single output F which is similar to what ice40 do? I am looking for a way to combine F LUT output and Q FF output.

gatecat, 2021-09-30:
> Can the GENERIC_SLICE in generic be modified into a single output F which is similar to what ice40 do? I am looking for a way to combine F LUT output and Q FF output.

You can have both as the same wire, the default packer will never use both at once. They were separate only to support arches where they are physically different wires.

Ravenslofty, 2022-02-15:
I suppose nowadays this could be achieved with the Viaduct framework.

gatecat, 2022-08-18:
> I suppose nowadays this could be achieved with the Viaduct framework.

Indeed, this is going to land as #1015

closed 2022-08-18 by gatecat