YosysHQ/nextpnr #740: interchange: Pre-binding site routing is fundamentally broken in more complex cases

gatecat, 2021-06-30:
Right now, the site router is used to pre-bind the routing on the inputs to (and, to some extent, outputs from) sites, as part of the various attempts to prevent the router from routing through sites using site routing (which isn't legal in the fpga-interchange format).

However, this means the site router is essentially picking input/output pins at random, on the assumption that all site pins are equally routeable. But there is no guarantee that this is the case - consider how the UltraScale+ implements its IOB sites:

[image]

`OP` _must_ be used as the input site pin to the output buffer unless this is the negative side of a pseudo-differential IOB, in which case `OUTB_B_IN` _must_ be used.

By coincidence, for the simple wire case, the site router wrongly picks `OUTB_B_IN` and the design is hence unrouteable.

I can see three ways to fix this:
 - Modify the site router to check routeability and only pick site pins that appear to be routeable (this will have to be on a heuristic that won't pick up every case, but should pick up most). This will probably be done for binding site routing only, at least to start with, to avoid the performance cost in other cases.
 - Disable input pre-binding for certain sites/site pins in the YAML config only.
 - Remove the input pre-binding altgother, doing site routing in the general router, and investigate other ways to prevent illegal site route throughs.

gatecat, 2021-06-30:
cc @acomodi, do you have any thoughts here?

it looks like you might have hit a similar case with the carry inputs in https://github.com/YosysHQ/nextpnr/blob/152c41c3ac4541ddfa2147be1ac89f9d0eaf5c6e/fpga_interchange/site_router.cc#L1103-L1134 ? unfortunately this doesn't generalise to other cases

acomodi, 2021-06-30:
Yeah, I think this issue is a more generic case than the carry chain one (and chains in general) for which the site router was selecting pins that are unroutable.

I think that the first solution might be the way to go. We may add a check that, in case a site pin is part of a dedicated interconnect (if there is no fast check we may pre-build a fast look-up) and check routeability only in that case, as those are the most critical ones as general interconnect cannot reach those site pins.

Moreover, it might make sense to add a detailed routing check only during the pre-router site router step, otherwise the check routeability check would run also when placing BELs.

gatecat, 2021-06-30:
Sounds good, I will work on that! I think we actually need to check three things:

 - pins that **have** to be used because of dedicated interconnect rules for the driving net
 - pins that can't be used because they are only driving dedicated interconnect (and that dedicated interconnect can't be used for the driving net)
 - pins that can't be used because they are not driven by any useful interconnect

PR #743 (2021-07-01): interchange: Reserve site ports only reachable from dedicated routing
Fixes #740, with this patch the "wire" test is now working on ZCU104 hardware

cc @acomodi
closed 2021-07-01 by gatecat