YosysHQ/nextpnr #753: [interchange] Global Logic nets start at tile pip

mtdudek, 2021-07-06:
Hi,

I'm working on STA in python-fpga-interchange.
I tried to pass physical netlist generated using command `make test-fpga_interchange-counter_arty35t-dcp` to STA.
It failed on GLOBAL_LOGIC0 as some sources begin at tile pip and not on belpin.
It looks like pip's input is connected directly to TIEOFF's HARD0 sitePin in INT_L tile, but nextpnr's netlist does not go as far as belpin inside TIEOFF site. It just starts from pip.

I think it's a bug.

Here is physical netlist generated with nextpnr in yaml format
[arty.txt](https://github.com/YosysHQ/nextpnr/files/6769347/arty.txt)

gatecat, 2021-07-06:
Would you be able to give an example of a specific route tree that's problematic inside that design?

mtdudek, 2021-07-06:
One the trees starts in line 9315 of arty.txt

gatecat, 2021-07-06:
My understanding is that `INT_L_X8Y45/GND_WIRE` is marked in Vivado, and consequently the interchange data too, as tied to ground. This means it's valid for a ground route to start there, ignoring the tieoff site. Handling this case in downstream tooling is necessary for UltraScale+ and Nexus support in any case, as neither have tie-off sites in the same way that 7-series does.

closed 2021-07-21 by mtdudek