YosysHQ/nextpnr #769: assertion failure instantiating SB_SPI on UP5K

fmahnke, 2021-07-20:
I'm trying to use this SPI example on a UP5K breakout board: https://github.com/damdoy/ice40_ultraplus_examples/tree/master/spi_hw

Trying to adapt it from arachne to nextpnr, I change the yosys/pnr commands in the Makefile to 

```
yosys -p "synth_ice40 -json $(filename).json -blif $(filename).blif" $(filename).v
nextpnr-ice40 --up5k --package sg48 --json $(filename).json --pcf $(pcf_file) --asc $(filename).asc
```

I get this error:

```
terminate called after throwing an instance of 'nextpnr_ice40::assertion_failure'
  what():  Assertion failure: is_string (/home/fritz/src/aur/nextpnr-git/src/nextpnr/common/property.h:83)
```

The problem seems to be the SBACKO and SO parameters of the SB_SPI  SB_SBI_inst in top.v . If those are commented out, nextpnr finishes normally.

Attached is the full output from yosys and nextpnr.

[out.txt](https://github.com/YosysHQ/nextpnr/files/6849161/out.txt)

gatecat, 2021-07-20:
Thanks for reporting! I think https://github.com/YosysHQ/nextpnr/pull/771 should fix the issue.

fmahnke, 2021-07-20:
I built and tested with the PR branch. Works now. :+1:

PR #771 (2021-07-20): ice40: Use default value when IP is missing BUS_ADDR74 parameter
Fixes #769
closed 2021-07-20 by gatecat