YosysHQ/nextpnr #773: [Interchange] XC7 OBUFDS misses a PIP begin enabled

mkurc-ant, 2021-07-23:
In https://github.com/SymbiFlow/prjxray/pull/1705 for the `OBUFDS` test the Nextpnr flow completes. But when a DCP file is open in Vivado it shows a missing route for the net that connects both parts of the `OBUFDS` cell. There is a PIP (a pseudo one) that connects them. See the screenshot:

[image]

mkurc-ant, 2021-07-23:
I have a suspicion what is happening: The device database defines macro expansion for:

- `OBUFTDS` -> `OBUFTDS_DUAL_BUF`
- `OBUFDS` -> `OBUFDS_DUAL_BUF`

But the latter is not defined in the macro library therefore nextpnr is not finding it. I've examined the output physical netlist and the `OBUFTDS` cell is not split there indeed. So this looks more like a RapidWright issue then.

gatecat, 2021-07-23:
have you tried with https://github.com/Xilinx/RapidWright/pull/203 ?

mkurc-ant, 2021-07-23:
Yes I have (its merged). I double checked. The `exceptionMap` is there and contains correct entries for both `OBUFTDS` and `OBUFDS`.

But then there is the macro library which defines only one of the expanded macros (this comes from a device file converted to JSON):
```
{
  "name": "OBUFTDS_DUAL_BUF",
  "propMap": {
    "entries": []
  },
  "view": "netlist",
  "lib": "macros",
  "ports": [
    {
      "_id": 2633
    },
    {
      "_id": 2634
    },
    {
      "_id": 2635
    },
    {
      "_id": 2636
    }
  ],
  "_id": 3027
}
```
and there is no such entry for `OBUFDS_DUAL_BUF`

gatecat, 2021-07-23:
Yeah, it really does look like it's missing from RapidWright (whereas it does exist in Vivado). The data file `hdi_primitives.edf.test` which appears to contain all the other macro expansions is missing it too.

mkurc-ant, 2021-08-24:
This is now fixed by using a patched version of RapidWright and will be fixed in its next release. Closing.

closed 2021-08-24 by mkurc-ant