YosysHQ/nextpnr #776: ECP5 how to put arrays into lpf file

rowanG077, 2021-07-24:
I have an ECP5 board and I want to use a sequence of input pins as an array. I have something likes this in my lpf:

```
LOCATE COMP "data[0]" SITE "J4";
IOBUF PORT "data[0]" IO_TYPE=LVCMOS33;
LOCATE COMP "data[1]" SITE "K3";
IOBUF PORT "data[1]" IO_TYPE=LVCMOS33;
LOCATE COMP "data[2]" SITE "G1";
IOBUF PORT "data[2]" IO_TYPE=LVCMOS33;
LOCATE COMP "data[3]" SITE "K4";
IOBUF PORT "data[3]" IO_TYPE=LVCMOS33;
LOCATE COMP "data[4]" SITE "C2";
IOBUF PORT "data[4]" IO_TYPE=LVCMOS33;
LOCATE COMP "data[5]" SITE "E3";
IOBUF PORT "data[5]" IO_TYPE=LVCMOS33;
```

Then in my top level verilog file I have this:

```
module Top
    ( ...
    , output wire [5:0] data
    );
```

This however doesn't work. The outputs pins are never driven. However when I change the lop and top to the following:

```
LOCATE COMP "data0" SITE "J4";
IOBUF PORT "data0" IO_TYPE=LVCMOS33;
LOCATE COMP "data1" SITE "K3";
IOBUF PORT "data1" IO_TYPE=LVCMOS33;
LOCATE COMP "data2" SITE "G1";
IOBUF PORT "data2" IO_TYPE=LVCMOS33;
LOCATE COMP "data3" SITE "K4";
IOBUF PORT "data3" IO_TYPE=LVCMOS33;
LOCATE COMP "data4" SITE "C2";
IOBUF PORT "data4" IO_TYPE=LVCMOS33;
LOCATE COMP "data5" SITE "E3";
IOBUF PORT "data5" IO_TYPE=LVCMOS33;
```

```
module Top
    ( ...
    , output wire data0
, output wire data0
, output wire data0
    );
```

gatecat, 2021-07-24:
Hi, @rowanG077! 

The former should work fine, please can you provide a more complete example of what's not working? Also, I note you say "input pins" but it's defined as an output in your design.

rowanG077, 2021-07-24:
@gatecat 

Yes indeed it works. I had a typo... sorry

closed 2021-07-24 by rowanG077