YosysHQ/nextpnr #781: ModuleNotFoundError: No module named 'database' when building bbasm on Debian Buster

polprog, 2021-07-27:
I am building nextpnr-ecp5 on buster. I have installed all the prerequisites, including project Trellis. However an error occurs when building:
```
$ cmake . -DARCH=ecp5 -DTRELLIS_INSTALL_PREFIX=/usr/local
[ runs fine ]
$ make 
[  3%] Built target bbasm
[  5%] Generating chipdb/chipdb-25k.bba
Traceback (most recent call last):
  File "/home/user/programming/lattice_fpga/nextpnr/ecp5/trellis_import.py", line 22, in <module>
    import database
ModuleNotFoundError: No module named 'database'
make[2]: *** [ecp5/CMakeFiles/chipdb-ecp5-bbas.dir/build.make:65: ecp5/chipdb/chipdb-25k.bba] Error 1
make[1]: *** [CMakeFiles/Makefile2:287: ecp5/CMakeFiles/chipdb-ecp5-bbas.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
```
Unfortunately the generic module name does not help, there is no module named `database` in pip3. Im running python3 version 3.7.3 and I confirmed that the offending script is ran with `python3` not `python`.

The value of `args.libdir` just before the failing import is `['/usr/lib/x86_64-linux-gnu/trellis', '/usr/local/share/trellis/util/common', '/usr/local/share/trellis/timing/util']`.

This prevents nextpnr from being built.

gatecat, 2021-07-27:
Can you run `ls /usr/local/share/trellis/util/common` and share the output?

polprog, 2021-07-27:
`/usr/local/share/trellis/util/common` does not exist.. as my Trellis install prefix is `/usr` not `/usr/local`. I didn't notice that. Now that I re-ran `cmake . -DARCH=ecp5 -DTRELLIS_INSTALL_PREFIX=/usr` with the correct prefix it builds fine!

Which raises the question why didn't the first cmake invocation report any errors? I think it would also be useful to standardise the prefix across the readmes of Trellis, yosys and nextpnr.

kittennbfive, 2021-08-03:
I had exactly the same problem on Debian 10 a few days ago. Looks like the path changed and the doc was not adjusted?

kittennbfive, 2021-08-28:
I  think - but i might be wrong! - that the documentation of prjtrellis needs an update. I created a PR: https://github.com/YosysHQ/prjtrellis/pull/177

gatecat, 2022-08-18:
With https://github.com/YosysHQ/prjtrellis/pull/177 the situation that led to this error should now be clarified.

closed 2022-08-18 by gatecat