YosysHQ/nextpnr #782: Nexus (LIFCL-40): ODDRX1 not supported

fkokosinski, 2021-07-28:
Hi, I'm experiencing the issue of the `ODDRX1` cell not being supported. The error I'm getting is as follows:
```
Info: Packing BRAM...
Info:     Created 28 OXIDE_EBR cells from:
Info:             12x PDP16K_MODE
Info:             16x PDPSC16K_MODE
Info: Packing LRAM...
Info:     Created 2 LRAM_CORE cells from:
Info:              2x SP512K_MODE
Info: Packing carries...
Info: Packing FFs...
Info:     Created 3384 OXIDE_FF cells from:
Info:             13x FD1P3BX
Info:           3276x FD1P3IX
Info:             95x FD1P3JX
ERROR: Cell type 'ODDRX1' instantiated as 'ODDRX1' is not supported by this device.
```

The instance of the `ODDRX1` cell generated by LiteX is as follows:
```
ODDRX1 ODDRX1(
	.D0(1'd0),
	.D1(1'd1),
	.SCLK(sys_clk),
	.Q(spiflash4x_analyzer)
);
```
I'm not getting this error when using Radiant, so I assume that this behavior is not intended.

tmichalak, 2021-07-28:
@gatecat is that somewhere on your priority list?

gatecat, 2021-07-28:
Yes, there's a bit more fuzzing work but it will be supported at some point. If there's a specific use case where this is important I'm happy to bump this up the priority list.

tmichalak, 2021-07-28:
Yes, that would be of great use as we are working on speeding up LiteSPI (we will open a PR shortly) to be used on Lattice NX boards in the [CFU Playground](https://github.com/google/CFU-Playground) project. For now our only option is to use Radiant.

gatecat, 2021-07-28:
I've now added basic support for this in #784, and given it a quick check on hardware! It would be great if you can test if this is working in your use case too - you'll need to update prjoxide to latest git master before rebuilding nextpnr.

fkokosinski, 2021-07-28:
@gatecat I've rebuilt prjoxide (HEAD on `c0c7208`) and then I've built nextpnr from your branch. While I'm no longer getting the previous error message, unfortunately now I'm getting the following exception when trying to build a design with the `ODDRX1` cell:
```
Info: Packing IOLOGIC...
Info:     Created 1 IOLOGIC cells from:
Info:              1x ODDRX1
terminate called after throwing an instance of 'nextpnr_nexus::assertion_failure'
  what():  Assertion failure: iol_bel != BelId() (/home/filip/Work/google-hps/nextpnr2/nexus/pack.cc:1222)
  ```

gatecat, 2021-07-28:
Very sorry, this one was my fault, I forgot to push my prjoxide changes! Please can you fetch & rebuild prjoxide again, then rebuild nextpnr and see if it works?

If not, please upload your design files somewhere and I'll have a look.

fkokosinski, 2021-07-28:
Thanks for the quick response. Unfortunately I'm still getting errors:
```
Info: Slack histogram:
Info:  legend: * represents 45 endpoint(s)
Info:          + represents [1,45) endpoint(s)
Info: [ -3769,    520) |********************+
Info: [   520,   4809) |************************************************************
Info: [  4809,   9098) |*************************************************+
Info: [  9098,  13387) |*******************************************+
Info: [ 13387,  17676) |
Info: [ 17676,  21965) |
Info: [ 21965,  26254) |
Info: [ 26254,  30543) |
Info: [ 30543,  34832) |
Info: [ 34832,  39121) |*+
Info: [ 39121,  43410) |+
Info: [ 43410,  47699) |
Info: [ 47699,  51988) |
Info: [ 51988,  56277) |
Info: [ 56277,  60566) |
Info: [ 60566,  64855) |
Info: [ 64855,  69144) |
Info: [ 69144,  73433) |
Info: [ 73433,  77722) |
Info: [ 77722,  82011) |+
Info: Checksum: 0x1091b0e9
Info: Running post-place optimisations...
Info:      created 933 direct LUT-FF pairs
Info: Routing globals...
ERROR: Failed to route net 'spiflash_phy_clk$glb_clk' from X0/Y28/JCLKO_DCC_DCC4 to X4/Y7/JM0_SLICEC using dedicated routing.
0 warnings, 1 error
```

I'm attaching the whole `gateware` directory that LiteX generates: [782_nextpnr_gateware.tar.gz](https://github.com/YosysHQ/nextpnr/files/6893709/782_nextpnr_gateware.tar.gz).

gatecat, 2021-07-28:
Thanks for reporting and including the testcase! This has turned out to be an unrelated issue with global clocks that also drive routeing. I've merged the I/ODDR support and created a new PR fixing this issue as https://github.com/YosysHQ/nextpnr/pull/785, do you want to give this a try?

Unfortunately it is failing timing currently, this is a known issue around routing congestion that I am working on more generally.

fkokosinski, 2021-07-29:
Thanks! I was able to build the design successfully and tested it on hardware. The `ODDRX1` cell is working correctly for me.

PR #784 (2021-07-28): nexus: Basic IDDRX1/ODDRX1 support
Fixes #782
closed 2021-07-29 by fkokosinski