YosysHQ/nextpnr #786: Add fmax reports for various clock domains in output (JSON?)

thoughtpolice, 2021-07-28:
When running nextpnr, I'd like to be able to iteratively run the P&R phase with randomized seeds, and look at how the fmax changes to probe for improvements. However, I don't see a current way to achieve this when using e.g. `nextpnr-ecp5 --write design.json` to write out a routed design.

Ideally I think the resulting format would just have A) a list of detected/inferred clocks (just their net names, I suppose), B) their target frequency C) the actual calculated fmax for each respective domain. Including the critical path would also be a really nice bonus.

This seems like it would be good to put into the `--write` output, but OTOH it seems as though that might become excessively large when you only want to read the resulting compilation report for a design that's big. (JSON parsers tend to be variously inefficient, and parsing a 1MB JSON file for 10 bytes of data is pointless.) So perhaps there should simply be a new, more general method for writing dedicated reports in machine readable format (perhaps with a schema).

**Q: Couldn't I just use the scripting API?** Normally I'd assume so, yes! But I'm using the **[YoWASP build](https://github.com/YoWASP/nextpnr) of nextpnr, where that support isn't available. So getting at this some other way is preferable.

gatecat, 2021-07-29:
Hi! I've currently added a basic report format in #787, also including utilisation info. Can you have a look and see if this suits your requirements - it's missing critical path info at the moment but I plan to look into this in the future.

thoughtpolice, 2021-07-30:
I've applied #787 to a personal build of nextpnr and tried it; that looks like the ticket! Even handles my PLLs as I expect. Thank you for the quick response! Maybe I can take a stab at adding a path for the critical path info...

cbalint13, 2021-07-30:
@thoughtpolice , @gatecat 

 If need to anneal max-clock you may want to look at: https://github.com/cbalint13/nextpnr-anneal

 * A future version (WiP) of annealer will look at partial runtimes (without letting  nextpnr to finish), extract feauture vectors  (numerics gathered till a early point) an will try to anneal it that way. Partial annealer is useful for large FPGA, but in this case final JSON report is useless (due that is final only).
  * I find it very easy to parse nextpnr  logs right  in realtime (can look at the code).

gatecat, 2022-08-18:
#787 added the basic reports, then #810 added some more detailed output

closed 2022-08-18 by gatecat