YosysHQ/nextpnr #792: Pin M9 is not working in Lattice ECP5 LFE5U-25F-6BG256C

benitoss, 2021-08-07:
I am using a Colorlight 5A-75E version 7.1 with bypassed their level shifters  (https://github.com/q3k/chubby75/blob/master/5a-75e/hardware_V7.1.md)
The FPGA is **Lattice ECP5 LFE5U-25F-6BG256C**

I am using the IceStudio + Apio solution --> https://icestudio.io/
But if you use the M9 pin I have got this error

**ERROR: IO pin 'vf69136$tr_io' constrained to pin 'M9', which does not exist for package 'CABGA256'.**

I suppose the error is in the toolchain of ECP5 because the rest of the pins are working perfectly

Why is the reason for this error ??

gatecat, 2021-08-07:
Hi! This should be working in the latest version of the toolchain, I think apio must be providing an older version where this pin wasn't supported.

benitoss, 2021-08-09:
Maybe Right, I have put in contact with the IceStudio & Apio creators to solve the problem.
Thanks

closed 2021-08-09 by benitoss