YosysHQ/nextpnr #796: Nexus/prjoxide: 'called `Option::unwrap()` on a `None` value'

tcal-x, 2021-08-16:
After updating cargo / rustc (to 1.54.0), I was able to rebuild prjoxide, but then when building nextpnr-nexus, I encountered:

```
$ make -j3
Scanning dependencies of target bbasm
Scanning dependencies of target chipdb-nexus-bbas
[  2%] Building CXX object bba/CMakeFiles/bbasm.dir/main.cc.o
[  4%] Generating chipdb/chipdb-LIFCL.bba
thread 'main' panicked at 'called `Option::unwrap()` on a `None` value', prjoxide/src/database.rs:481:62
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
make[2]: *** [nexus/CMakeFiles/chipdb-nexus-bbas.dir/build.make:82: nexus/chipdb/chipdb-LIFCL.bba] Error 101
make[1]: *** [CMakeFiles/Makefile2:278: nexus/CMakeFiles/chipdb-nexus-bbas.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[  6%] Linking CXX executable bbasm
[  6%] Built target bbasm
make: *** [Makefile:149: all] Error 2
```
With full backtrace:
```
$ RUST_BACKTRACE=full make -j3
[  2%] Generating chipdb/chipdb-LIFCL.bba
[  6%] Built target bbasm
thread 'main' panicked at 'called `Option::unwrap()` on a `None` value', prjoxide/src/database.rs:481:62
stack backtrace:
   0:     0x55b9e5654b30 - std::backtrace_rs::backtrace::libunwind::trace::h34055254b57d8e79
                               at /rustc/a178d0322ce20e33eac124758e837cbd80a6f633/library/std/src/../../backtrace/src/backtrace/libunwind.rs:90:5
   1:     0x55b9e5654b30 - std::backtrace_rs::backtrace::trace_unsynchronized::h8f1e3fbd9afff6ec
                               at /rustc/a178d0322ce20e33eac124758e837cbd80a6f633/library/std/src/../../backtrace/src/backtrace/mod.rs:66:5
   2:     0x55b9e5654b30 - std::sys_common::backtrace::_print_fmt::h3a99a796b770c360
                               at /rustc/a178d0322ce20e33eac124758e837cbd80a6f633/library/std/src/sys_common/backtrace.rs:67:5
   3:     0x55b9e5654b30 - <std::sys_common::backtrace::_print::DisplayBacktrace as core::fmt::Display>::fmt::h32d1f94a80615d18
                               at /rustc/a178d0322ce20e33eac124758e837cbd80a6f633/library/std/src/sys_common/backtrace.rs:46:22
   4:     0x55b9e567540c - core::fmt::write::h306731c068f7162c
                               at /rustc/a178d0322ce20e33eac124758e837cbd80a6f633/library/core/src/fmt/mod.rs:1110:17
   5:     0x55b9e56529c5 - std::io::Write::write_fmt::hd2fa90334eee2a21
                               at /rustc/a178d0322ce20e33eac124758e837cbd80a6f633/library/std/src/io/mod.rs:1588:15
   6:     0x55b9e5656aab - std::sys_common::backtrace::_print::h5abaa2601a852287
                               at /rustc/a178d0322ce20e33eac124758e837cbd80a6f633/library/std/src/sys_common/backtrace.rs:49:5
   7:     0x55b9e5656aab - std::sys_common::backtrace::print::h8d81445442bb638f
                               at /rustc/a178d0322ce20e33eac124758e837cbd80a6f633/library/std/src/sys_common/backtrace.rs:36:9
   8:     0x55b9e5656aab - std::panicking::default_hook::{{closure}}::hcfe804496a9fa747
                               at /rustc/a178d0322ce20e33eac124758e837cbd80a6f633/library/std/src/panicking.rs:208:50
   9:     0x55b9e5656581 - std::panicking::default_hook::hbea8e3ccf2ba8901
                               at /rustc/a178d0322ce20e33eac124758e837cbd80a6f633/library/std/src/panicking.rs:225:9
  10:     0x55b9e5657174 - std::panicking::rust_panic_with_hook::h7ee9e1a2d0f8975a
                               at /rustc/a178d0322ce20e33eac124758e837cbd80a6f633/library/std/src/panicking.rs:622:17
  11:     0x55b9e5656c27 - std::panicking::begin_panic_handler::{{closure}}::h8ab3b4491718b2c7
                               at /rustc/a178d0322ce20e33eac124758e837cbd80a6f633/library/std/src/panicking.rs:517:13
  12:     0x55b9e565502c - std::sys_common::backtrace::__rust_end_short_backtrace::hd489062ffa586a9f
                               at /rustc/a178d0322ce20e33eac124758e837cbd80a6f633/library/std/src/sys_common/backtrace.rs:141:18
  13:     0x55b9e5656bb9 - rust_begin_unwind
                               at /rustc/a178d0322ce20e33eac124758e837cbd80a6f633/library/std/src/panicking.rs:515:5
  14:     0x55b9e551efd1 - core::panicking::panic_fmt::hca6330e3e14086b4
                               at /rustc/a178d0322ce20e33eac124758e837cbd80a6f633/library/core/src/panicking.rs:92:14
  15:     0x55b9e551ef1d - core::panicking::panic::h1a48d878ff3dcd40
                               at /rustc/a178d0322ce20e33eac124758e837cbd80a6f633/library/core/src/panicking.rs:50:5
  16:     0x55b9e55b19b0 - prjoxide::database::Database::new_builtin::h15c4e2b03bb02aba
  17:     0x55b9e5530369 - prjoxide::main::h575ba15bc3404e31
  18:     0x55b9e5525a33 - std::sys_common::backtrace::__rust_begin_short_backtrace::h07fb99eb4133c7b4
  19:     0x55b9e5525a49 - std::rt::lang_start::{{closure}}::h1332c40f95e49a3c
  20:     0x55b9e5657679 - core::ops::function::impls::<impl core::ops::function::FnOnce<A> for &F>::call_once::h2aabc384aab89b7b
                               at /rustc/a178d0322ce20e33eac124758e837cbd80a6f633/library/core/src/ops/function.rs:259:13
  21:     0x55b9e5657679 - std::panicking::try::do_call::hc5fcacb7a85fc7b1
                               at /rustc/a178d0322ce20e33eac124758e837cbd80a6f633/library/std/src/panicking.rs:401:40
  22:     0x55b9e5657679 - std::panicking::try::hb5d9603af3abbe3a
                               at /rustc/a178d0322ce20e33eac124758e837cbd80a6f633/library/std/src/panicking.rs:365:19
  23:     0x55b9e5657679 - std::panic::catch_unwind::h98fe6ac3925e64b4
                               at /rustc/a178d0322ce20e33eac124758e837cbd80a6f633/library/std/src/panic.rs:434:14
  24:     0x55b9e5657679 - std::rt::lang_start_internal::h22ac7383c516f93e
                               at /rustc/a178d0322ce20e33eac124758e837cbd80a6f633/library/std/src/rt.rs:34:21
  25:     0x55b9e55324e2 - main
  26:     0x7f6163a570b3 - __libc_start_main
  27:     0x55b9e551f7ae - _start
  28:                0x0 - <unknown>
make[2]: *** [nexus/CMakeFiles/chipdb-nexus-bbas.dir/build.make:82: nexus/chipdb/chipdb-LIFCL.bba] Error 101
make[1]: *** [CMakeFiles/Makefile2:278: nexus/CMakeFiles/chipdb-nexus-bbas.dir/all] Error 2
make: *** [Makefile:149: all] Error 2
```
fyi @gatecat ; maybe this is transient due to other changes in flight?

gatecat, 2021-08-16:
Unfortunately I can't reproduce the failure here. It seems like it could be caused by an empty/broken database submodule in prjoxide, can you check that was checked out correctly? Otherwise we'll have to look into rustc version issues.

tcal-x, 2021-08-16:
I had needed to update cargo since before, prjoxide wouldn't build due to an issue with "clap".

I'll try starting from scratch with everything.

tcal-x, 2021-08-16:
@gatecat , I'm supposed to use master for both prjoxide and nextpnr, right?

tcal-x, 2021-08-16:
Ok, it works now.   Possibly before I'd forgotten to init/update the submodules in prjoxide before building.   Thanks for checking on it @gatecat.

closed 2021-08-16 by tcal-x