YosysHQ/nextpnr #797: Fails to build on Alpine v3.14 due to fortify/stdio.h:70:28: error: inlining failed in call to 'always_inline' 'vsnprintf'

natevw, 2021-08-17:
After working around https://github.com/YosysHQ/nextpnr/issues/411#issuecomment-900525718 I am still unable to build nextpnr under Alpine Linux. It finishes compiling but hits an error as it goes to link things:

```
[100%] Linking CXX executable nextpnr-ice40
/usr/include/c++/10.3.1/ext/string_conversions.h: In function '__to_xstring.constprop':
/usr/include/fortify/stdio.h:70:28: error: inlining failed in call to 'always_inline' 'vsnprintf': function body can be overwritten at link time
70 | _FORTIFY_FN(vsnprintf) int vsnprintf(char *__s, size_t __n, const char *__f,
| ^
/usr/include/c++/10.3.1/ext/string_conversions.h:111:32: note: called from here
111 | const int __len = __convf(__s, __n, __fmt, __args);
| ^
make[3]: *** [/tmp/ccPdOpEJ.mk:119: /tmp/nextpnr-ice40.pbbCBc.ltrans39.ltrans.o] Error 1
make[3]: *** Waiting for unfinished jobs....
lto-wrapper: fatal error: make returned 2 exit status
compilation terminated.
/usr/lib/gcc/x86_64-alpine-linux-musl/10.3.1/../../../../x86_64-alpine-linux-musl/bin/ld: error: lto-wrapper failed
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/nextpnr-ice40.dir/build.make:773: nextpnr-ice40] Error 1
make[1]: *** [CMakeFiles/Makefile2:203: CMakeFiles/nextpnr-ice40.dir/all] Error 2
make: *** [Makefile:136: all] Error 2
The command '/bin/sh -c cd nextpnr; Eigen3_DIR=/usr/share/cmake/Modules cmake -DARCH=ice40 -DCMAKE_INSTALL_PREFIX=/usr/local . && make -j$(nproc) && make install' returned a non-zero code: 2
```

This seems to be related to https://gitlab.alpinelinux.org/alpine/aports/-/issues/8626 where the recommended solution appears to be:

> fix your source code to work with fortify_source. [i.e. Alpine's `-D_FORTIFY_SOURCE=2` compile flag?]

There's some other potentially-related workarounds discussed at e.g. https://github.com/ClickHouse/clickhouse-odbc/issues/12#issuecomment-574791054 and other projects, but I'm not sure what the best path forward would be for this project specifically.

Is there something that can/should be changed in whatever generates `nextpnr-ice40.pbbCBc.ltrans39.ltrans.o` or is this a bug in Alpine's `c++/10.3.1/ext/string_conversions.h` itself?

natevw, 2021-08-17:
Here's my Dockerfile so far if it's helpful to repro:

```
# HT: https://github.com/suyashkumar/icestorm-docker/blob/master/Dockerfile
# HT: https://github.com/ddm/icetools/blob/master/Dockerfile
# cf. http://www.clifford.at/icestorm/ [still the official link, via https://github.com/YosysHQ/icestorm#readme?] 

FROM alpine:3.14
RUN apk add --no-cache --virtual runtime-dependencies \
libgcc libstdc++ libffi boost libftdi1 readline \
graphviz python3 perl tcl make bash

RUN apk add --no-cache --virtual build-dependencies \
git mercurial clang build-base autoconf cmake \
bison flex flex-dev gawk tcl-dev libffi-dev boost-dev \
python3-dev libftdi1-dev readline-dev eigen-dev gperf

RUN git clone --depth 1 https://github.com/YosysHQ/icestorm.git icestorm
RUN cd icestorm; make -j$(nproc) && make install
RUN rm -rf icestorm

# TODO: https://gitlab.alpinelinux.org/alpine/aports/-/issues/8626
RUN git clone --depth 1 https://github.com/YosysHQ/nextpnr nextpnr
RUN cd nextpnr; Eigen3_DIR=/usr/share/cmake/Modules cmake -DARCH=ice40 -DCMAKE_INSTALL_PREFIX=/usr/local . && make -j$(nproc) && make install
RUN rm -rf nextpnr

RUN git clone --depth 1 https://github.com/YosysHQ/yosys.git yosys
RUN cd yosys; make -j$(nproc) && make install
RUN rm -rf yosys

RUN apk del --purge build-dependencies
```

gatecat, 2021-08-20:
Looking at some other discussions around this issue, disabling IPO (passing `-DUSE_IPO=off` to CMake) should probably be a reasonable workaround. It's unclear if there's anything actionable on the nextpnr side unfortunately.

josuah, 2023-08-02:
> disabling IPO (passing `-DUSE_IPO=off` to CMake) should probably be a reasonable workaround

This workaround works (Alpine Linux 3.18.0)!

Tested with:
- nextpnr-ecp5-master
- nextpnr-gowin-master
- nextpnr-ice40-master
- nextpnr-nexus-master

closed 2024-09-26 by gatecat