YosysHQ/nextpnr #803: Strange linker error on macOS

modwizcode, 2021-08-24:
Trying to build with the GUI or without on macOS with support for the ice40 (don't think GUI or target is related there).

Have an issue, at the end get this:
```
[ 92%] Building CXX object CMakeFiles/nextpnr-ice40.dir/ice40/bitstream.cc.o
[ 93%] Building CXX object CMakeFiles/nextpnr-ice40.dir/ice40/cells.cc.o
[ 94%] Building CXX object CMakeFiles/nextpnr-ice40.dir/ice40/chains.cc.o
[ 95%] Building CXX object CMakeFiles/nextpnr-ice40.dir/ice40/delay.cc.o
[ 96%] Building CXX object CMakeFiles/nextpnr-ice40.dir/ice40/gfx.cc.o
[ 97%] Building CXX object CMakeFiles/nextpnr-ice40.dir/ice40/main.cc.o
[ 98%] Building CXX object CMakeFiles/nextpnr-ice40.dir/ice40/pack.cc.o
4 warnings generated.
[ 99%] Building CXX object CMakeFiles/nextpnr-ice40.dir/ice40/pcf.cc.o
[100%] Linking CXX executable nextpnr-ice40
ld: library not found for -ltbb
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [nextpnr-ice40] Error 1
make[1]: *** [CMakeFiles/nextpnr-ice40.dir/all] Error 2
make: *** [all] Error 2
```

`libtbb` is findable with a normal `pkg-config` invocation and the files seem to be there, so don't really know what's going on here.

gatecat, 2021-08-24:
tbb was included to speed up some part of the fpga interchange arch that isn't actually used, I'll look at getting it disabled by default to eliminate this class of problem, I think.

modwizcode, 2021-08-24:
This issue seems to be that when TBB gets added to the list of libraries, it's not referenced correctly to the package, so it just adds a blank dependency. This means it's not actually picking up any modified paths it seems. Which is probably the issue on macOS.

https://github.com/YosysHQ/nextpnr/blob/6fc41692d689beda2f142ab039d7fa8d2517bebe/CMakeLists.txt#L260 

It can be fixed by changing `tbb` to `TBB::tbb` on that line

PR #971 (2022-04-06): cmake: properly include TBB libraries.
This fixes #803 by properly following the libraries for TBB. I have not been able to test this against anything other than macOS and Linux.
closed 2022-04-06 by gatecat