YosysHQ/nextpnr #811: Lattice ECP5: ERROR: cell type 'mux21' is unsupported

SiliconWizard, 2021-09-13:
Trying to use the MUX21 primitive.

Is this unsupported on purpose?
Is there a workaround?

gatecat, 2021-09-14:
There isn't really a generic useful MUX2 in hardware. I expect in almost all cases the Lattice tools just turn it into a LUT implementing MUX functions.

The hardware mux primitives are PFUMX and L6MUX21 but these are only intended for use in LUT-MUX structures and not on their own.

SiliconWizard, 2021-09-14:
OK thanks. I suppose that implementing 'MUX2' in pure HDL should yield the same hardware?

Lattice IPs use the MUX2 primitive a lot, and I'm actually trying to port one for use with Yosys/nextpnr-ecp5.

gatecat, 2021-09-14:
Yeah, it should

SiliconWizard, 2021-09-14:
Thanks - I found this page that lists all supported primitives, hoping it's up to date:
https://github.com/YosysHQ/nextpnr/blob/master/ecp5/docs/primitives.md

closed 2022-05-21 by Ravenslofty