YosysHQ/nextpnr #818: Lattice ECP5 LFE5U-25F-6BG256C: Pin M9 is not working

yodalee, 2021-09-20:
I am using the ECP5 develop board [icesugar-pro](https://github.com/wuxx/icesugar-pro) and the FPGA is `LFE5U-25F-6BG256C`.
I tried to connect to the SPI-FLASH on this board. According to the [schematic](https://github.com/wuxx/icesugar-pro/blob/master/schematic/iCESugar-Pro-v1.3.pdf), the FLASH_CLK is connected to port `N9`.

So I set the lpf file as below:
```
# top.lpf SPI-FLASH
LOCATE COMP "flash_cs"   SITE "N8";
LOCATE COMP "flash_mosi" SITE "T8";
LOCATE COMP "flash_miso" SITE "T7";
LOCATE COMP "flash_io2"  SITE "M7";
LOCATE COMP "flash_io3"  SITE "N7";
LOCATE COMP "flash_clk"  SITE "N9";

IOBUF PORT "flash_cs"   IO_TYPE=LVCMOS33 DRIVE=4;
IOBUF PORT "flash_clk"  IO_TYPE=LVCMOS33 DRIVE=4;
IOBUF PORT "flash_mosi" IO_TYPE=LVCMOS33 DRIVE=4;
IOBUF PORT "flash_miso" IO_TYPE=LVCMOS33 DRIVE=4;
IOBUF PORT "flash_io2"  IO_TYPE=LVCMOS33 DRIVE=4;
IOBUF PORT "flash_io3"  IO_TYPE=LVCMOS33 DRIVE=4;
```

and compile my project with following command:
```
nextpnr-ecp5 --25k --package CABGA256 --speed 6 --json top.json --textcfg top_out.config --lpf top.lpf
```

However I will get this error:
```
ERROR: IO pin 'flash_clk$tr_io' constrained to pin 'N9', which does not exist for package 'CABGA256'.
ERROR: Packing design failed.
```

I have check the [prjtrellis database](https://github.com/YosysHQ/prjtrellis-db), which in the [pio.json file of ECP5 LFE5U-25F](https://github.com/YosysHQ/prjtrellis-db/blob/master/ECP5/LFE5U-25F/iodb.json) do have no port `N9`.

Is this a setting error? or I just cannot use port N9 to communicate with the flash?

gatecat, 2021-09-20:
You  need to use USRMCLK for this pin, rather than using it as a regular IO.

yodalee, 2021-09-20:
Thank for your reply. Here is what I understand after search the keyword of USRMCLK.
1. Just remove the `LOCATE COMP` and `IOBUF PORT` definition of `flash_clk` in lpf file.
2. In top.sv, I can access the clock with `USRMCLK` module
```systemverilog
module top (
  /* SPI-FLASH */
  output logic flash_cs,
  output logic flash_mosi,
  input flash_miso,
  input flash_io2,
  input flash_io3,
);

logic tristate = 1'b0;
USRMCLK u1(.USRMCLKI(flash_clk), .USRMCLKTS(tristate));
```
Now I can control the clk to flash by adjust the wire tristate.
Am I right?

gatecat, 2021-09-20:
Yeah that should work

yodalee, 2021-09-20:
Thanks, let me do some testing.

closed 2021-09-20 by yodalee