YosysHQ/nextpnr #819: Yet another Linux Python Boost related build failure: PopOS edition

sedurCode, 2021-09-20:
I had this doozy yesterday when building nextpnr as part of the '[summon_fpga_tools](https://github.com/open-tool-forge/summon-fpga-tools)' script put together for the icebreaker fpga board. I was building on a fresh (and updated) install of PoPOS with functioning python3.9 and dependencies installed. 

`Traceback (most recent call last):
  File "/home/<my_path_to_build_script>/summon-fpga-tools/nextpnr-e926cddca274423ea414e61a07deb785ee63c92d/ecp5/trellis_import.py", line 21, in <module>
    import pytrellis
ImportError: /lib/x86_64-linux-gnu/libboost_python39.so.1.74.0: undefined symbol: _Py_NewReference
make[2]: *** [ecp5/CMakeFiles/chipdb-ecp5-bbas.dir/build.make:84: ecp5/chipdb/chipdb-25k.bba] Error 1
make[1]: *** [CMakeFiles/Makefile2:659: ecp5/CMakeFiles/chipdb-ecp5-bbas.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....`

A switft google didn't yield a simple response / fix for this issue. 
Any suggestions on how to continue?

gatecat, 2021-09-20:
Can you try pointing this to `YosysHQ/prjtrellis` instead https://github.com/open-tool-forge/summon-fpga-tools/blob/master/summon-fpga-tools.sh#L349 ? prjtrellis moved a while ago and it looks like sft hasn't been updated.

sedurCode, 2021-09-22:
Yep, that worked a treat! 
Thanks @gatecat

closed 2021-09-22 by sedurCode