YosysHQ/nextpnr #820: nextpnr-nexus apparent hang at iteration #1

tcal-x, 2021-09-21:
Command:
```
nextpnr-nexus --json hps_proto2_platform.json --pdc hps_proto2_platform.pdc --fasm hps_proto2_platform.fasm     --device LIFCL-17-8UWG72C   --seed 1
```
Version (although I can reproduce with every version I've tried):
```
nextpnr-nexus -- Next Generation Place and Route (Version 6fc41692)
```

Complete output:
```
Info: constraining clock net 'por_clk' to 41.25 MHz
Info: Constraining LVCMOS18H IO 'user_led0_OB_O' to pin G3 (PCLKT3_2/ADC_CP13; bel X66/Y29/PIOA)
Info: Constraining LVCMOS18 IO 'spiflash_wp_IB_I' to pin B3 (MD2; bel X63/Y0/PIOB)
Info: Constraining LVCMOS18 IO 'spiflash_mosi_BB_B' to pin B5 (MOSI/MD0; bel X61/Y0/PIOB)
Info: Constraining LVCMOS18 IO 'spiflash_miso_BB_B' to pin C4 (MISO/MD1; bel X63/Y0/PIOA)
Info: Constraining LVCMOS18 IO 'spiflash_hold_IB_I' to pin B2 (MD3; bel X65/Y0/PIOA)
Info: Constraining LVCMOS18 IO 'spiflash_cs_n_OB_O' to pin A3 (MCSN/PCLKT0_1; bel X61/Y0/PIOA)
Info: Constraining LVCMOS18 IO 'spiflash_clk_OB_O' to pin B4 (MCLK/PCLKT0_0; bel X59/Y0/PIOA)
Info: Constraining LVCMOS18H IO 'serial_tx_OB_O' to pin G1 (LRC_GPLL0C_IN/VREF3_2; bel X70/Y29/PIOB)
Info: Constraining LVCMOS18 IO 'serial_rx_IB_I' to pin E2 (EIO; bel X75/Y9/PIOB)
Info: Packing IOLOGIC...
Info: Packing DSPs...
Info: Packing BRAM...
Info:     Created 14 OXIDE_EBR cells from:
Info:             14x PDPSC16K_MODE
Info: Packing LRAM...
Info:     Created 5 LRAM_CORE cells from:
Info:              5x SP512K_MODE
Info: Packing carries...
Info: Packing FFs...
Info:     Created 4454 OXIDE_FF cells from:
Info:              7x FD1P3BX
Info:           4356x FD1P3IX
Info:             91x FD1P3JX
Info: Packing LUTs...
Info: Promoting globals...
Info:      promoting clock net 'por_clk'
Info: Placing globals...
Info:     constraining OSCA 'OSC_CORE' to bel 'X71/Y0/OSC_CORE'
Info:     constraining por_clk$glb_clk$drv_DCC 'DCC' to bel 'X37/Y0/DCC_T1' based on dedicated routing
Info: Generating derived timing constraints...
Warning:     Overriding derived constraint of 41.3 MHz on net por_clk with user-specified constraint of 37.5 MHz.
Info:     Derived frequency constraint of 41.3 MHz for net por_clk$glb_clk

Info: Annotating ports with timing budgets for target frequency 12.00 MHz
Info: Checksum: 0x4be3281e

Info: Device utilisation:
Info:               OXIDE_FF:  4454/13824    32%
Info:             OXIDE_COMB:  9542/13824    69%
Info:                   RAMW:    60/ 1728     3%
Info:            SEIO33_CORE:     7/   23    30%
Info:            SEIO18_CORE:     2/   44     4%
Info:          DIFFIO18_CORE:     0/   22     0%
Info:               OSC_CORE:     1/    1   100%
Info:              OXIDE_EBR:    14/   24    58%
Info:           PREADD9_CORE:    32/   48    66%
Info:             MULT9_CORE:    32/   48    66%
Info:            MULT18_CORE:     8/   24    33%
Info:             REG18_CORE:    28/   48    58%
Info:         MULT18X36_CORE:     2/   12    16%
Info:            MULT36_CORE:     1/    6    16%
Info:             ACC54_CORE:     0/   12     0%
Info:                    DCC:     1/   62     1%
Info:                VCC_DRV:     1/   74     1%
Info:              LRAM_CORE:     5/    5   100%
Info:                IOLOGIC:     0/   44     0%
Info:               SIOLOGIC:     0/   22     0%

Info: Placed 11 cells based on constraints.
Info: Creating initial analytic placement for 11513 cells, random placement wirelen = 585928.
Info:     at initial placer iter 0, wirelen = 6092
Info:     at initial placer iter 1, wirelen = 5699
Info:     at initial placer iter 2, wirelen = 5613
Info:     at initial placer iter 3, wirelen = 5732
Info: Running main analytical placer.
Info:     at iteration #1, type OXIDE_COMB: wirelen solved = 13442, spread = 452433, legal = 452420; time = 0.19s
```

At this point I see no further output for 15 minutes, although it continues using 100% of a core.

Files attached, including a README.

[iter1hang.zip](https://github.com/YosysHQ/nextpnr/files/7205938/iter1hang.zip)

gatecat, 2021-09-21:
Thanks, this looks to be an issue around DSP placement. I will investigate tomorrow.

gatecat, 2021-09-22:
I think https://github.com/YosysHQ/nextpnr/pull/821 should be a fix for this.

tcal-x, 2021-09-22:
> I think #821 should be a fix for this.

@gatecat , I confirmed that it fixes my case.

tcal-x, 2021-09-22:
Fixed by #821 .

closed 2021-09-22 by tcal-x