YosysHQ/nextpnr #831: Optimization when nextpnr-ice40 >100% device utilization

regymm, 2021-09-29:
I'm new to nextpnr(and open source FPGA toolchains at all) so apologies if the problem has already been asked before. 

I'm porting a piece of program from the EPM1270 CPLD(1270 logic elements) to iCE40-HX1K FPGA. On Quartus II, after synthesis, the total logic element usage is 1447, which is more than 1270. Then after place & route is run, the logic element usage is optimized into 1125/1270, and can successfully fit into the CPLD. 

However, when using yosys and nextpnr toolchain with the same program for the HX1K, after yosys, nextpnr reported 1403/1280 ICESTORM_LC usage then failed instantly. So I wonder are there options that allow nextpnr to try harder to optimize in this case? (In smaller designs I tried, yosys/nextpnr used less resources on HX1K than Quartus on EPM1270, so I guess the 1403 LCs can certainly be optimized to fit in HX1k. And my program uses no BRAM). 

These are the full output of nextpnr and statistics of yosys: 

```
Warning: No PCF file specified; IO pins will be placed automatically

Info: Packing constants..
Info: Packing IOs..
Info: Packing LUT-FFs..
Info:      896 LCs used as LUT4 only
Info:      245 LCs used as LUT4 and DFF
Info: Packing non-LUT FFs..
Info:      121 LCs used as DFF only
Info: Packing carries..
Info:       97 LCs used as CARRY only
Info: Packing RAMs..
Info: Placing PLLs..
Info: Packing special functions..
Info: Packing PLLs..
Info: Promoting globals..
Info: promoting clk$SB_IO_IN (fanout 368)
Info: promoting $abc$29675$auto$rtlil.cc:1969:NotGate$29515 [reset] (fanout 32)
Info: promoting $abc$29675$auto$rtlil.cc:1969:NotGate$29429 [reset] (fanout 16)
Info: promoting $abc$29675$auto$rtlil.cc:1969:NotGate$29467 [reset] (fanout 16)
Info: promoting $abc$29675$auto$rtlil.cc:1969:NotGate$29579 [reset] (fanout 16)
Info: promoting $abc$29675$auto$dff2dffe.cc:175:make_patterns_logic$25167 [cen] (fanout 28)
Info: promoting $abc$29675$auto$dff2dffe.cc:175:make_patterns_logic$26949 [cen] (fanout 23)
Info: Constraining chains...
Info:       42 LCs used to legalise carry chains.
Info: Checksum: 0x41606025

Info: Annotating ports with timing budgets for target frequency 12.00 MHz
Info: Checksum: 0xbe47a392

Info: Device utilisation:
Info: 	         ICESTORM_LC:  1403/ 1280   109%
Info: 	        ICESTORM_RAM:     1/   16     6%
Info: 	               SB_IO:    51/  112    45%
Info: 	               SB_GB:     7/    8    87%
Info: 	        ICESTORM_PLL:     0/    1     0%
Info: 	         SB_WARMBOOT:     0/    1     0%

Info: Placed 0 cells based on constraints.
Info: Creating initial analytic placement for 1096 cells, random placement wirelen = 16779.
Info:     at initial placer iter 0, wirelen = 616
Info:     at initial placer iter 1, wirelen = 656
Info:     at initial placer iter 2, wirelen = 695
Info:     at initial placer iter 3, wirelen = 684
Info: Running main analytical placer.
ERROR: Failed to expand region (0, 0) |_> (13, 17) of 1403 ICESTORM_LCs
1 warning, 1 error
```

```
5.44. Printing statistics.

=== top ===

   Number of wires:               1230
   Number of wire bits:           2674
   Number of public wires:         168
   Number of public wire bits:     968
   Number of memories:               0
   Number of memory bits:            0
   Number of processes:              0
   Number of cells:               1787
     SB_CARRY                      279
     SB_DFF                         68
     SB_DFFE                       166
     SB_DFFESR                      48
     SB_DFFESS                       2
     SB_DFFSR                       82
     SB_LUT4                      1141
     SB_RAM40_4K                     1
```

gatecat, 2021-10-01:
Different ways of estimating logic cell counts and splitting optimisations means there's no guarantee that the split between synthesis and place and route will be the same for Quartus and Yosys+nextpnr.

This is particularly the case for different architectures with different carry logic styles; packing rules; etc.

That said, if you can point me to the design in question I can have a look for any obvious problems.

regymm, 2021-10-01:
Thanks for solving my confusion. Then it seems the CPLD just does better at implementing a very large state machine.

closed 2021-10-01 by regymm