YosysHQ/nextpnr #832: Compiling on a Cygwin fails

71GA, 2021-09-30:
I am trying to compile nextpnr on a Cygwin:

┌───┐
│ $ │ ziga > windows--workstation > ~
└─┬─┘
  └─> git clone https://github.com/YosysHQ/nextpnr
Cloning into 'nextpnr'...
remote: Enumerating objects: 23777, done.
remote: Counting objects: 100% (3186/3186), done.
remote: Compressing objects: 100% (1170/1170), done.
remote: Total 23777 (delta 2130), reused 2886 (delta 1979), pack-reused 20591
Receiving objects: 100% (23777/23777), 10.09 MiB | 1.18 MiB/s, done.
Resolving deltas: 100% (17166/17166), done.

┌───┐
│ $ │ ziga > windows--workstation > ~
└─┬─┘
  └─> cd nextpnr/

┌───┐
│ $ │ ziga > windows--workstation > nextpnr
└─┬─┘
  └─> cmake . -DARCH=ice40 -DPYTHON_INCLUDE_DIR=/usr/include/python3.8  -DPYTHON_LIBRARY=/usr/lib                                                       -- The CXX compiler identification is GNU 11.2.0
-- The C compiler identification is GNU 11.2.0
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Building with IPO
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.8.10", minimum required is "3.5")
-- Found PythonLibs: /usr/lib (found suitable version "3.8.10", minimum required is "3.5")
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE
-- Found Boost: /usr/include (found version "1.66.0") found components: filesystem program_options iostreams system thread regex chrono date_time atomic
-- Found Boost: /usr/include (found version "1.66.0") found components: program_options filesystem system
-- Configuring architecture: ice40
-- Enabled iCE40 devices: 384;1k;5k;u4k;8k
-- IceStorm install prefix: /usr/local
-- icebox data directory: /usr/local/share/icebox
-- Using iCE40 chipdb: /home/ziga/nextpnr/ice40/chipdb
-- Configuring done
WARNING: Target "nextpnr-ice40" requests linking to directory "/usr/lib".  Targets may link only to libraries.  CMake is dropping the item.
-- Generating done
-- Build files have been written to: /home/ziga/nextpnr

┌───┐
│ $ │ ziga > windows--workstation > nextpnr
└─┬─┘
  └─> make -j$(nproc)
[  1%] Building CXX object bba/CMakeFiles/bbasm.dir/main.cc.o
[  3%] Generating chipdb/chipdb-384.bba
[  4%] Generating chipdb/chipdb-1k.bba
[  6%] Linking CXX executable bbasm.exe
[  6%] Built target bbasm
[  7%] Generating chipdb/chipdb-5k.bba
[  9%] Generating chipdb/chipdb-u4k.bba
[ 10%] Generating chipdb/chipdb-8k.bba
[ 10%] Built target chipdb-ice40-bbas
[ 12%] Generating ice40/chipdb/chipdb-384.cc
[ 13%] Generating ice40/chipdb/chipdb-1k.cc
[ 15%] Generating ice40/chipdb/chipdb-5k.cc
[ 16%] Generating ice40/chipdb/chipdb-u4k.cc
[ 18%] Generating ice40/chipdb/chipdb-8k.cc
[ 18%] Built target chipdb-ice40-bins
[ 20%] Building CXX object CMakeFiles/chipdb-ice40.dir/ice40/chipdb/chipdb-1k.cc.o
[ 21%] Building CXX object CMakeFiles/chipdb-ice40.dir/ice40/chipdb/chipdb-384.cc.o
[ 23%] Building CXX object CMakeFiles/chipdb-ice40.dir/ice40/chipdb/chipdb-5k.cc.o
[ 24%] Building CXX object CMakeFiles/chipdb-ice40.dir/ice40/chipdb/chipdb-u4k.cc.o
[ 26%] Building CXX object CMakeFiles/chipdb-ice40.dir/ice40/chipdb/chipdb-8k.cc.o
[ 33%] Built target chipdb-ice40
[ 35%] Building CXX object CMakeFiles/nextpnr-ice40.dir/common/archcheck.cc.o
[ 36%] Building CXX object CMakeFiles/nextpnr-ice40.dir/common/basectx.cc.o
[ 38%] Building CXX object CMakeFiles/nextpnr-ice40.dir/common/bits.cc.o
[ 40%] Building CXX object CMakeFiles/nextpnr-ice40.dir/common/command.cc.o
[ 41%] Building CXX object CMakeFiles/nextpnr-ice40.dir/common/context.cc.o
[ 43%] Building CXX object CMakeFiles/nextpnr-ice40.dir/common/design_utils.cc.o
[ 44%] Building CXX object CMakeFiles/nextpnr-ice40.dir/common/embed.cc.o
[ 46%] Building CXX object CMakeFiles/nextpnr-ice40.dir/common/handle_error.cc.o
[ 47%] Building CXX object CMakeFiles/nextpnr-ice40.dir/common/idstring.cc.o
[ 49%] Building CXX object CMakeFiles/nextpnr-ice40.dir/common/idstringlist.cc.o
[ 50%] Building CXX object CMakeFiles/nextpnr-ice40.dir/common/log.cc.o
/home/ziga/nextpnr/common/log.cc: In function ‘std::string nextpnr_ice40::vstringf(const char*, va_list)’:
/home/ziga/nextpnr/common/log.cc:75:9: error: ‘vasprintf’ was not declared in this scope; did you mean ‘vsprintf’?
   75 |     if (vasprintf(&str, fmt, ap) < 0)
      |         ^~~~~~~~~
      |         vsprintf
make[2]: *** [CMakeFiles/nextpnr-ice40.dir/build.make:216: CMakeFiles/nextpnr-ice40.dir/common/log.cc.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/Makefile2:203: CMakeFiles/nextpnr-ice40.dir/all] Error 2
make: *** [Makefile:136: all] Error 2

What is with the `vasprintf’`? Where can I get this function?

gatecat, 2021-10-01:
Hi! Can you see if https://github.com/YosysHQ/nextpnr/pull/834 fixes your problem as I don't have a Cygwin env immediately available to test?

71GA, 2021-10-01:
@gatecat So I should use a fixed makefile fto compile yosys and hopefully this will fix my problem with nextpnr?

71GA, 2021-10-01:
If I use your makefile I get this errror:

```
┌───┐
│ $ │ ziga > windows--workstation > yosys
└─┬─┘
  └─> make
[  0%] Building kernel/version_62739f7b.cc
[  0%] Building kernel/version_62739f7b.o
clang-8: error: unsupported option '-fPIC' for target 'x86_64-unknown-windows-cygnus'
make: *** [Makefile:501: kernel/version_62739f7b.o] Error 1
```

If I modify makefile i.e. remove parameter `-fPIC` for the `CXXFLAGS` it prints another error:

```
┌───┐
│ $ │ ziga > windows--workstation > yosys
└─┬─┘
  └─> make
[  0%] Building kernel/version_62739f7b.cc
[  0%] Building kernel/version_62739f7b.o
[  0%] Building kernel/driver.o
[  0%] Building techlibs/common/simlib_help.inc
/bin/sh: techlibs/common/cellhelp.py: Permission denied
make: *** [techlibs/common/Makefile.inc:12: techlibs/common/simlib_help.inc] Error 126
```

gatecat, 2021-10-01:
No, nothing to do with Yosys or Makefiles - its the patch to log.cc in that pull request.

71GA, 2021-10-01:
@gatecat Ah! Sorry! I clicked your link and then the "top link" there... Ok I will test immediately!

71GA, 2021-10-01:
@gatecat Compilation now proceeds, gets to 100% but then experiences another error right before the end:

```
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/nextpnr-ice40.dir/build.make:772: nextpnr-ice40.exe] Error 1
make[1]: *** [CMakeFiles/Makefile2:203: CMakeFiles/nextpnr-ice40.dir/all] Error 2
make: *** [Makefile:136: all] Error 2
```

gatecat, 2021-10-01:
Can you provide a few more lines of context?

71GA, 2021-10-01:
OK! I just compile again! We will have to wait a bit...

71GA, 2021-10-01:
This is the entire procedure:

```
┌───┐
│ $ │ ziga > windows--workstation > nextpnr
└─┬─┘
  └─> make -j$(nproc)
[  1%] Generating chipdb/chipdb-384.bba
[  3%] Building CXX object bba/CMakeFiles/bbasm.dir/main.cc.o
[  4%] Generating chipdb/chipdb-1k.bba
[  6%] Linking CXX executable bbasm.exe
[  6%] Built target bbasm
[  7%] Generating chipdb/chipdb-5k.bba
[  9%] Generating chipdb/chipdb-u4k.bba
[ 10%] Generating chipdb/chipdb-8k.bba
[ 10%] Built target chipdb-ice40-bbas
[ 12%] Generating ice40/chipdb/chipdb-384.cc
[ 13%] Generating ice40/chipdb/chipdb-1k.cc
[ 15%] Generating ice40/chipdb/chipdb-5k.cc
[ 16%] Generating ice40/chipdb/chipdb-u4k.cc
[ 18%] Generating ice40/chipdb/chipdb-8k.cc
[ 18%] Built target chipdb-ice40-bins
[ 21%] Building CXX object CMakeFiles/chipdb-ice40.dir/ice40/chipdb/chipdb-1k.cc.o
[ 21%] Building CXX object CMakeFiles/chipdb-ice40.dir/ice40/chipdb/chipdb-384.cc.o
[ 23%] Building CXX object CMakeFiles/chipdb-ice40.dir/ice40/chipdb/chipdb-5k.cc.o
[ 24%] Building CXX object CMakeFiles/chipdb-ice40.dir/ice40/chipdb/chipdb-u4k.cc.o
[ 26%] Building CXX object CMakeFiles/chipdb-ice40.dir/ice40/chipdb/chipdb-8k.cc.o
[ 33%] Built target chipdb-ice40
[ 35%] Building CXX object CMakeFiles/nextpnr-ice40.dir/common/archcheck.cc.o
[ 36%] Building CXX object CMakeFiles/nextpnr-ice40.dir/common/basectx.cc.o
[ 38%] Building CXX object CMakeFiles/nextpnr-ice40.dir/common/bits.cc.o
[ 40%] Building CXX object CMakeFiles/nextpnr-ice40.dir/common/command.cc.o
[ 41%] Building CXX object CMakeFiles/nextpnr-ice40.dir/common/context.cc.o
[ 43%] Building CXX object CMakeFiles/nextpnr-ice40.dir/common/design_utils.cc.o
[ 44%] Building CXX object CMakeFiles/nextpnr-ice40.dir/common/embed.cc.o
[ 46%] Building CXX object CMakeFiles/nextpnr-ice40.dir/common/handle_error.cc.o
[ 47%] Building CXX object CMakeFiles/nextpnr-ice40.dir/common/idstring.cc.o
[ 49%] Building CXX object CMakeFiles/nextpnr-ice40.dir/common/idstringlist.cc.o
[ 50%] Building CXX object CMakeFiles/nextpnr-ice40.dir/common/log.cc.o
[ 52%] Building CXX object CMakeFiles/nextpnr-ice40.dir/common/nextpnr.cc.o
[ 53%] Building CXX object CMakeFiles/nextpnr-ice40.dir/common/nextpnr_assertions.cc.o
[ 55%] Building CXX object CMakeFiles/nextpnr-ice40.dir/common/nextpnr_namespaces.cc.o
[ 56%] Building CXX object CMakeFiles/nextpnr-ice40.dir/common/nextpnr_types.cc.o
[ 58%] Building CXX object CMakeFiles/nextpnr-ice40.dir/common/place_common.cc.o
[ 60%] Building CXX object CMakeFiles/nextpnr-ice40.dir/common/placer1.cc.o
[ 61%] Building CXX object CMakeFiles/nextpnr-ice40.dir/common/placer_heap.cc.o
[ 63%] Building CXX object CMakeFiles/nextpnr-ice40.dir/common/property.cc.o
[ 64%] Building CXX object CMakeFiles/nextpnr-ice40.dir/common/pybindings.cc.o
[ 66%] Building CXX object CMakeFiles/nextpnr-ice40.dir/common/report.cc.o
[ 67%] Building CXX object CMakeFiles/nextpnr-ice40.dir/common/router1.cc.o
[ 69%] Building CXX object CMakeFiles/nextpnr-ice40.dir/common/router2.cc.o
[ 70%] Building CXX object CMakeFiles/nextpnr-ice40.dir/common/sdf.cc.o
[ 72%] Building CXX object CMakeFiles/nextpnr-ice40.dir/common/str_ring_buffer.cc.o
[ 73%] Building CXX object CMakeFiles/nextpnr-ice40.dir/common/svg.cc.o
[ 75%] Building CXX object CMakeFiles/nextpnr-ice40.dir/common/timing.cc.o
[ 76%] Building CXX object CMakeFiles/nextpnr-ice40.dir/common/timing_opt.cc.o
[ 78%] Building CXX object CMakeFiles/nextpnr-ice40.dir/3rdparty/json11/json11.cpp.o
[ 80%] Building CXX object CMakeFiles/nextpnr-ice40.dir/json/jsonwrite.cc.o
[ 81%] Building CXX object CMakeFiles/nextpnr-ice40.dir/frontend/json_frontend.cc.o
[ 83%] Building CXX object CMakeFiles/nextpnr-ice40.dir/ice40/arch.cc.o
[ 84%] Building CXX object CMakeFiles/nextpnr-ice40.dir/ice40/arch_place.cc.o
[ 86%] Building CXX object CMakeFiles/nextpnr-ice40.dir/ice40/arch_pybindings.cc.o
[ 87%] Building CXX object CMakeFiles/nextpnr-ice40.dir/ice40/bitstream.cc.o
[ 89%] Building CXX object CMakeFiles/nextpnr-ice40.dir/ice40/cells.cc.o
[ 90%] Building CXX object CMakeFiles/nextpnr-ice40.dir/ice40/chains.cc.o
[ 92%] Building CXX object CMakeFiles/nextpnr-ice40.dir/ice40/delay.cc.o
[ 93%] Building CXX object CMakeFiles/nextpnr-ice40.dir/ice40/gfx.cc.o
[ 95%] Building CXX object CMakeFiles/nextpnr-ice40.dir/ice40/main.cc.o
[ 96%] Building CXX object CMakeFiles/nextpnr-ice40.dir/ice40/pack.cc.o
[ 98%] Building CXX object CMakeFiles/nextpnr-ice40.dir/ice40/pcf.cc.o
[... 2 lines cut ...]
/usr/lib/gcc/x86_64-pc-cygwin/11/../../../../x86_64-pc-cygwin/bin/ld: CMakeFiles/nextpnr-ice40.dir/common/basectx.cc.o (symbol from plugin):(.gnu.linkonce.t._ZN5boost16exception_detail10clone_implINS0_19error_info_injectorINS_17bad_function_callEEEED1Ev[_ZThn16_N5boost16exception_detail10clone_implINS0_19error_info_injectorINS_17bad_function_callEEEED1Ev]+0x0): multiple definition of `non-virtual thunk to boost::exception_detail::clone_impl<boost::exception_detail::error_info_injector<boost::bad_function_call> >::~clone_impl()'; CMakeFiles/nextpnr-ice40.dir/common/archcheck.cc.o (symbol from plugin):(.gnu.linkonce.t._ZN5boost16exception_detail10clone_implINS0_19error_info_injectorINS_17bad_function_callEEEED1Ev[_ZTv0_n40_N5boost16exception_detail10clone_implINS0_19error_info_injectorINS_17bad_function_callEEEED1Ev]+0x0): first defined here
/usr/lib/gcc/x86_64-pc-cygwin/11/../../../../x86_64-pc-cygwin/bin/ld: CMakeFiles/nextpnr-ice40.dir/common/basectx.cc.o (symbol from plugin):(.gnu.linkonce.t._ZN5boost16exception_detail10clone_implINS0 [... cut ...] les/nextpnr-ice40.dir/common/archcheck.cc.o (symbol from plugin):(.gnu.linkonce.t._ZN5boost16exception_detail10clone_implINS0_14bad_exception_EED0Ev[_ZThn40_N5boost16exception_detail10clone_implINS0_14bad_exception_EED0Ev]+0x0): first defined here
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/nextpnr-ice40.dir/build.make:773: nextpnr-ice40.exe] Error 1
make[1]: *** [CMakeFiles/Makefile2:203: CMakeFiles/nextpnr-ice40.dir/all] Error 2
make: *** [Makefile:136: all] Error 2
```

gatecat, 2021-10-01:
I expect this is related to https://github.com/YosysHQ/nextpnr/issues/684. Can you try adding `-DUSE_IPO=OFF` to the CMake command line and building again?

71GA, 2021-10-01:
It worked.

71GA, 2021-10-01:
So in order to compile on Cygwin, you have to install some Cygwin packages: `cmake`, `python38`, `python38-devel`, `libboost-devel`, `eigen3`. and then do this: 
```
git clone https://github.com/YosysHQ/nextpnr
cd nextpnr/
```
aAfter this is done, we replace `/common/log.cc` with the patched version, ythat @gatecat provided. After this is done we can compile:
```
cmake . -DARCH=ice40 -DPYTHON_INCLUDE_DIR=/usr/include/python3.8  -DPYTHON_LIBRARY=/usr/lib/libpython3.8.dll.a -DUSE_IPO=OFF
make -j$(nproc)
make install
```
I wrote this in case anyone needs.

gatecat, 2021-10-01:
Thanks for writing that up, for the record now the pull request is merged the patch to `common/log.cc` is no longer required.

closed 2021-10-01 by gatecat

71GA, 2021-10-01:
**Note:**
Maybe we also need packages: `boost-build`, `libboost_python3-devel`, `mingw64-x86_64-boost`, `python3-boost.mpi`. About this I am not sure.