YosysHQ/nextpnr #835: Compiling on Debian 11 fails

christian-krieg, 2021-10-01:
Hey there,

While nextpnr was compiling like a charm on Debian 10, it unfortunately fails with Debian 11. I did the following:

Install pre-requisites:

```
sudo apt-get install build-essential clang bison flex libreadline-dev gawk tcl-dev libffi-dev git mercurial graphviz xdot pkg-config python python3 libftdi-dev python3-dev libboost-all-dev cmake libeigen3-dev qtbase5-dev qtchooser qt5-qmake qtbase5-dev-tools
```

Install Project Icestorm:
```
git clone https://github.com/YosysHQ/icestorm.git icestorm
cd icestorm
make -j$(nproc)
sudo make install
cd ..
```

Install nextpnr:
```git clone https://github.com/YosysHQ/nextpnr nextpnr
cd nextpnr
cmake -DARCH=ice40 -DCMAKE_INSTALL_PREFIX=/usr/local -DICESTORM_INSTALL_PREFIX=/usr/local .
make -j$(nproc)
```

CMake issues a bunch of warnings, notifying about a newer `Boost` version:
```
CMake Warning at /usr/share/cmake-3.18/Modules/FindBoost.cmake:1187 (message):
  New Boost version may have incorrect or missing dependencies and imported
  targets
Call Stack (most recent call first):
  /usr/share/cmake-3.18/Modules/FindBoost.cmake:1311 (_Boost_COMPONENT_DEPENDENCIES)
  /usr/share/cmake-3.18/Modules/FindBoost.cmake:1919 (_Boost_MISSING_DEPENDENCIES)
  bba/CMakeLists.txt:4 (find_package)

-- Found Boost: /usr/include (found version "1.74.0") found components: program_options filesystem system 
```

`make` aborts with the following error:
```
[  4%] Generating chipdb/chipdb-384.bba
[  4%] Building CXX object bba/CMakeFiles/bbasm.dir/main.cc.o
[  4%] Generating chipdb/chipdb-1k.bba
[  4%] Linking CXX executable bbasm
[  4%] Built target bbasm
[  4%] Generating chipdb/chipdb-5k.bba
[  4%] Generating chipdb/chipdb-u4k.bba
[  4%] Generating chipdb/chipdb-8k.bba
Killed
make[2]: *** [ice40/CMakeFiles/chipdb-ice40-bbas.dir/build.make:118: ice40/chipdb/chipdb-8k.bba] Error 137
make[1]: *** [CMakeFiles/Makefile2:278: ice40/CMakeFiles/chipdb-ice40-bbas.dir/all] Error 2
make: *** [Makefile:149: all] Error 2
```

Could you please provide a pointer? That would be awesome! :)

Cheers,
Christian

gatecat, 2021-10-01:
Is it possible that you are running out of memory, it looks a lot like a OOM kill to me.

christian-krieg, 2021-10-01:
Oh, baby, yes, that made my day, and definitely my weekend! :)

Thank you so much for your quick reply! Never thought it could be that easy. I was installing into a virtual machine, so 2G turned out to be too small. (With 8G it worked very smoothly, although `htop`  shows memory usage never significantly over 1G).

Have a good one!

closed 2021-10-01 by christian-krieg