YosysHQ/nextpnr #856: FF and DELAYG fails to pack into IOLOGIC

gregdavill, 2021-11-03:
## Observed behaviour

When trying to combine FF and DELAYG element into an IOLOGIC block packing fails
```
ERROR: Failed to pack flipflop 'OFS1P3BX' with 'syn_useioff' set into IOLOGIC.
ERROR: Packing design failed.
```

## Example
`test.v`
```verilog
module packer_fail(
   input wire clk,
    input wire _in,
    output wire _out
);

wire in_signal;

OFS1P3BX OFS1P3BX(
	.D(_in),
	.PD(1'd0),
	.SCLK(clk),
	.SP(1'd1),
	.Q(in_signal)
);

DELAYG #(
	.DEL_MODE("USER_DEFINED"),
	.DEL_VALUE(6'd60)
) DELAYG (
	.A(in_signal),
	.Z(_out)
);

endmodule
```
`test.lfp`
```
LOCATE COMP "_in" SITE "L18";
LOCATE COMP "clk" SITE "F18";
LOCATE COMP "_out" SITE "G18";
```

Run the following to build
```
$ yosys -p 'read_verilog test.v; synth_ecp5 -json test.json -top packer_fail'
$ nextpnr-ecp5 --json test.json --lpf test.lpf --textcfg test.config --25k --package CSFBGA285
```

## Expected result

Design packs successfully

gatecat, 2021-11-05:
Thanks, https://github.com/YosysHQ/nextpnr/pull/857 should hopefully fix this. If you are able to do so I'd appreciate a quick test on hardware otherwise I'll set something up to test.

gregdavill, 2021-11-05:
Thanks @gatecat 

Yes this patch is working for me!

PR #857 (2021-11-05): ecp5: Fix packing of IOFF with IODELAYs
Fixes #856
closed 2021-11-05 by gatecat