YosysHQ/nextpnr #860: Failed to route from PCSCLKDIV primitive on ecp5

BlindQlouder, 2021-11-10:
Hi, I am trying to divide a clock from the SERDES DCU using the module PCSCLKDIV. Nextpnr-ecp5 produces:

Warning: Failed to find a route for arc 0 of net tx_clk_div.
ERROR: Routing design failed.

[Here is a minimal example](https://github.com/BlindQlouder/nextpnr-pcsclkdiv-issue)

With Lattice Diamond a similar project compiles fine.

My nextpnr-ecp5 version is b7207b08

Thanks.

madscientist159, 2023-01-03:
I'm seeing the same issue -- any progress? :smile:

FeepingCreature, 2023-02-04:
Still happens on 48b0025732f7b5b1b7b091691bb5a26634ae5069

FeepingCreature, 2023-02-04:
I'm not sure if this is useful, but I used a reducer to bring it down to this JSON file:

```
{
  "modules": {
    "top": {
      "cells": {
        "17": {
          "type": "TRELLIS_FF",
          "port_directions": {
            "CLK": "input",
            "DI": "input"
          },
          "connections": {
            "CLK": [ 4 ],
            "DI": [ 54 ]
          }
        },
        "inst": {
          "type": "PCSCLKDIV",
          "port_directions": {
            "CDIVX": "output"
          },
          "connections": {
            "CDIVX": [ 4 ]
          }
        }
      },
      "netnames": {
        "tx_clk_div": {
          "bits": [ 4 ]
        }
      }
    }
  }
}
```
Where `nextpnr-ecp5 --json top.json --um5g-25k --package CABGA381` reproduces the error.

Unfortunately I don't know the format, so I don't know if this file is even still valid.

TheZoq2, 2023-12-15:
I have also hit this bug and I'm a little bit tempted to try to fix it. Are there any pointers to where I should start searching for what the problem and/or solution is?