YosysHQ/nextpnr #875: TypeError when building ECP5

grvvy, 2021-12-15:
System info:
```
Ubuntu 20.04.3
clang version 10.0.0-4ubuntu1 
Target: x86_64-pc-linux-gnu
Thread model: posix
InstalledDir: /usr/bin
gcc (Ubuntu 9.3.0-17ubuntu1~20.04) 9.3.0
```

I installed project trellis and yosys from source and checked prerequisites before trying to build for ECP5 with no success:

```
$ cmake -DARCH=ecp5 -DTRELLIS_INSTALL_PREFIX=/usr .
-- Building with IPO
-- Found Boost: /usr/include (found version "1.71.0") found components: filesystem program_options iostreams system thread regex chrono date_time atomic 
-- Found Boost: /usr/include (found version "1.71.0") found components: program_options filesystem system 
-- Configuring architecture: ecp5
-- Enabled ECP5 devices: 25k;45k;85k
-- Trellis install prefix: /usr
-- Trellis library directory: /usr/lib/x86_64-linux-gnu/trellis
-- Trellis data directory: /usr/share/trellis
-- Using ECP5 chipdb: /home/username/repos/nextpnr/ecp5/chipdb
-- Configuring done
-- Generating done
-- Build files have been written to: /home/username/repos/nextpnr
username@systemname:~/repos/nextpnr$ make -j$(nproc)
[  1%] Generating chipdb/chipdb-25k.bba
[  5%] Built target bbasm
Traceback (most recent call last):
  File "/home/username/repos/nextpnr/ecp5/trellis_import.py", line 640, in <module>
    main()
  File "/home/username/repos/nextpnr/ecp5/trellis_import.py", line 628, in main
    ddrg = pytrellis.make_dedup_chipdb(chip, include_lutperm_pips=True)
TypeError: make_dedup_chipdb(): incompatible function arguments. The following argument types are supported:
    1. (arg0: pytrellis.Chip) -> pytrellis.DedupChipdb

Invoked with: <pytrellis.Chip object at 0x7f48110900b0>; kwargs: include_lutperm_pips=True
make[2]: *** [ecp5/CMakeFiles/chipdb-ecp5-bbas.dir/build.make:65: ecp5/chipdb/chipdb-25k.bba] Error 1
make[1]: *** [CMakeFiles/Makefile2:259: ecp5/CMakeFiles/chipdb-ecp5-bbas.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
```

gatecat, 2021-12-15:
What version of trellis did you install and where did you get it from?

grvvy, 2021-12-15:
Ah, I was using a copied cmake command from prjtrellis: `cmake -DARCH=ecp5 -DTRELLIS_INSTALL_PREFIX=/usr/local .` and didn't notice the difference when checking the nextpnr readme: `cmake . -DARCH=ecp5 -DTRELLIS_INSTALL_PREFIX=/usr/local`

Using the one from the readme here worked.

closed 2021-12-15 by grvvy